wpilibsuite/allwpilib #4154: Rename HolonomicDriveController.calculate() arguments to be more descriptive

calcmogul, 2022-04-08:
It's currently ambiguous which pose should be used for what. `poseRef` is the trajectory pose. The heading is used to determine the velocity vector of the trajectory. `angleRef` is the desired chassis heading (the trajectory spline heading and chassis heading can be different for holonomic drivetrains). Perhaps `poseRef` should be renamed to `trajectoryPose`.

carelesshippo, 2022-04-08:
If these were changed also changing the names in the `RamseteController` would be a good idea to keep it somewhat consistent.

calcmogul, 2022-04-08:
They don't control the same thing though. With RamseteController, `poseRef` _is_ actually a pose reference.

carelesshippo, 2022-04-08:
Oh. I guess that makes the name change even more necessary than.

PR #4683 (2022-11-24): Rename HolonomicDriveController.calculate params
Fixes #4154
closed 2022-11-24 by PeterJohnson