wpilibsuite/allwpilib #4180: The system cannot find the path specified // Error: Could not find or load main class org.gradle.wrapper.GradleWrapperMain

IdlePhoenix1645, 2022-04-24:
**Describe the question you have.**
How do I fix a certain message I get when building my java project
**Describe the reason for your confusion.**
I keep getting an error message when building my java project and I don't know how to fix it.

**Is your question related to a problem? Please describe.**
Yes, I keep getting A specific set of error messages when I build my java project 
**Additional context**
The terminal returns the following output below when I try to build my project 

The system cannot find the path specified
Error: Could not find or load main class org.gradle.wrapper.GradleWrapperMain
Caused by: java.lang.ClassNotFoundException: org.gradle.wrapper.GradleWrapperMain
The terminal process "cmd.exe /d /c gradlew build -Dorg.gradle.java.home="c:users\public\wpilib\2022\jdk"" terminated with exit code 1.

Daltz333, 2022-04-24:
Have you verified you are in the correct project directory? It is likely that essential gradle files are missing. Can you create a new project and reproduce this?

IdlePhoenix1645, 2022-04-24:
I tried to simulate the examples and they started up just fine, but my project won't start the simulation

IdlePhoenix1645, 2022-04-24:
I created a new project and it builds fine? I don't know what I did to make my initial project fail to build

Daltz333, 2022-04-24:
I would recommend just copying your code over to the new project.

IdlePhoenix1645, 2022-04-24:
I will do that, I am just trying to figure out why it happened

ThadHouse, 2022-04-24:
Did you happen to push your code to git? That error is caused by the jar in gradle/wrapper missing. We’ve seen cases where teams remove that file from the .gitignore, and then builds afterwards fail with that error.

IdlePhoenix1645, 2022-04-24:
No, I have seen people talk about that. I haven't used git for my projects. I have also seen people talk about OneDrive causing issues, but I do not have OneDrive set up on this computer

Daltz333, 2022-04-24:
Some extremely aggressive anti-viruses could possibly remove it too.

IdlePhoenix1645, 2022-04-24:
Would Windows defender be considered extremely aggressive?  I've found that it likes to yell at me about trusting folders when I make new projects.

Daltz333, 2022-04-24:
I would not think it would be Windows defender. Have you moved the project files around at all in the past?

IdlePhoenix1645, 2022-04-24:
I moved the entire project folder to desktop, but I never moved a  Folder/file around inside the project folder

Daltz333, 2022-04-24:
Just a conjecture, but it's possible that the move only partially moved the folder. A folder move is simply a collection of individual file moves.

IdlePhoenix1645, 2022-04-24:
That is odd but it does make sense

IdlePhoenix1645, 2022-04-24:
I copied my code over into a new project and now I am receiving the error messages again @Daltz333

prateekma, 2022-07-27:
Are you still dealing with this issue? @IdlePhoenix1645

IdlePhoenix1645, 2022-07-27:
I haven't had access to my coding laptop for months, I cannot confirm or deny if the issue is still there at this moment.

Starlight220, 2023-01-22:
Can this be closed?