wpilibsuite/allwpilib #4191: Tracer epochs for subsystem periodics are named differently in C++

Starlight220, 2022-04-28:
There's a `Tracer` epoch for each subsystem periodic, and in Java it's named with the subsystem's name (ie `subsystem.getClass().getSimpleName()`) but in C++ the name is hardcoded to `"Subsystem Periodic()"`; maybe this is due to C++ not having runtime reflection?

calcmogul, 2022-04-28:
Yes, that's why.

Starlight220, 2022-04-28:
The command epochs use `GetTypeName`, so if that works then why can't it be used here as well? Or does `GetTypeName` not work either?

calcmogul, 2022-04-28:
There's GetName() which calls GetTypeName(), but I'm not sure whether it returns the name of the base class or the derived class.

Starlight220, 2022-04-28:
```cpp
#include <iostream>
#include <typeinfo>

class A {
    public:
    virtual ~A() = default;
};

class B : public A {};

int main()
{
    A a{};
    B b{};
    
    A* ptr{&a};
    std::cout<<typeid(*ptr).name() << "\n";
    ptr = &b;
    std::cout<<typeid(*ptr).name();

    return 0;
}
```
prints this:
```
1A
1B
```
so I guess that as long as there's a virtual method then it checks the derived class. I don't know what that `1` prefix is about though.

virtuald, 2022-04-28:
@Starlight220 did you check for if this worked for all platforms for debug and release? pybind11 only retrieves names of types for error messages when compiled for debug, not sure why that is (might be performance)

You can use this to demangle the names: https://github.com/pybind/pybind11/blob/ec24786eab4d170f21536c92422a59959e14dde4/include/pybind11/detail/typeid.h#L34

calcmogul, 2022-04-28:
We already have this function in Command.h, which is what should be getting called instead of typeid() directly.
```
template <typename T>
std::string GetTypeName(const T& type) {
  return wpi::Demangle(typeid(type).name());
}
```

PR #5836 (2023-12-01): [cmd] Add GetName to Subsystem, use in Scheduler tracer epochs
Continuation from #4199 
Fixes #4191

My only concern is calling typeid in C++ may be expensive (as mentioned in #4199). Ideally, Subsystem handles generating the name, rather than CommandScheduler caching it. The problem is that Subsystem can't access the subclass' type info at construction time. Any ideas would be helpful