wpilibsuite/allwpilib #4234: [tracking] Command-based improvements

Starlight220, 2022-05-14:
This refers to specifically commands themselves, not the button bindings stuff (that's in #4218).
All this is subject to change and happily up for discussion.

- [ ] #4179 

- [ ] #3608 

- [ ] command factory methods

- [ ] `unless` decorator (creates a `ConditionalCommand` with a null branch)

- [ ] way to break out of a sequential group from inside it

- [ ] `onInterrupt` decorator for command (groups)? 
       - would subsume #4021 

- [ ] (possibly?) user-defined infix decorators

- Anything else?

shueja-personal, 2022-05-14:
Another suggestion, related to the "unless"": Interrupt decorators only check the condition in `isFinished()`, so if I have a command that I want to skip completely if the condition is true, but also end if the condition becomes true while the command is running, I need this awkward construction: 
```java
new ConditionalCommand(
   new InstantCommand(),
   command.withInterrupt(condition),
   condition
)
```
This would seem to be a generally useful behavior.

shueja-personal, 2022-05-15:
Command factory method issue: #4236

Daltz333, 2022-05-15:
Another thing. Any PR that seeks to close this issue should also make corresponding PR(s) to [frc-docs](https://github.com/wpilibsuite/frc-docs).

closed 2022-10-23 by Starlight220