wpilibsuite/allwpilib #4237: Separate issue, I think. It's probably this:

Starlight220, 2022-05-15:
Separate issue, I think. It's probably this:
```cpp
enum AxisType { kXAxis, kYAxis, kZAxis, kTwistAxis, kThrottleAxis };
```

_Originally posted by @calcmogul in https://github.com/wpilibsuite/allwpilib/pull/4230#issuecomment-1126987665_

calcmogul, 2022-05-15:
Here's the relevant HAL documentation.
```cpp
/**
 * Gets the type of a specific joystick axis.
 *
 * This is device specific, and different depending on what system input type
 * the joystick uses.
 *
 * @param joystickNum the joystick number
 * @param axis        the axis number
 * @return the enumerated axis type
 */
int32_t HAL_GetJoystickAxisType(int32_t joystickNum, int32_t axis);
```

PeterJohnson, 2022-05-15:
Untested with actual joysticks, but it appears to be:
`enum { X, Y, Z, Rx, Ry, Rz, Slider, POV, None }`
Where Slider and POV are device specific:

0.    X-axis, usually the left-right movement of a stick.

1.    Y-axis, usually the forward-backward movement of a stick.

2.    Z-axis, often the throttle control. If the joystick does not have this axis, the value is 0.

3.    X-axis rotation. If the joystick does not have this axis, the value is 0.

4.    Y-axis rotation. If the joystick does not have this axis, the value is 0.

5.    Z-axis rotation (often called the rudder).  If the joystick does not have this axis, the value is 0.

6-7.  Two additional axes, formerly called the u-axis and v-axis, whose semantics depend on the joystick.