wpilibsuite/allwpilib #4263: Using development build fails with missing cscore object file

Sheraz4201, 2022-05-23:
**Describe the question you have.**
A clear and concise description of what you want clarification on.
When trying to use the development build, the code would not run and we found that our riolog returned an error of the cscore object file missing. I've left a copy of the riolog below with the errors. 

**Additional context**
java.io.IOException: cscorejni could not be loaded from path or an embedded resource.
attempted to load for platform /linux/athena/
Last Load Error:
/usr/local/frc/third-party/lib/libcscorejni.so: libwpinet.so: cannot open shared object file: No such file or directory
at edu.wpi.first.util.RuntimeLoader.loadLibrary(RuntimeLoader.java:94)
at edu.wpi.first.cscore.CameraServerJNI.<clinit>(CameraServerJNI.java:37)
at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:406)
at frc.robot.Main.main(Main.java:23)
java.io.IOException: cscorejni could not be loaded from path or an embedded resource.
attempted to load for platform /linux/athena/
Last Load Error:
/usr/local/frc/third-party/lib/libcscorejni.so: libwpinet.so: cannot open shared object file: No such file or directory
at edu.wpi.first.util.RuntimeLoader.loadLibrary(RuntimeLoader.java:94)
at edu.wpi.first.cscore.CameraServerJNI.<clinit>(CameraServerJNI.java:37)
at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:406)
at frc.robot.Main.main(Main.java:23)
java.io.IOException: cscorejni could not be loaded from path or an embedded resource.
attempted to load for platform /linux/athena/
Last Load Error:
/usr/local/frc/third-party/lib/libcscorejni.so: libwpinet.so: cannot open shared object file: No such file or directory
at edu.wpi.first.util.RuntimeLoader.loadLibrary(RuntimeLoader.java:94)
at edu.wpi.first.cscore.CameraServerJNI.<clinit>(CameraServerJNI.java:37)
at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:406)
at frc.robot.Main.main(Main.java:23)
java.io.IOException: cscorejni could not be loaded from path or an embedded resource.
attempted to load for platform /linux/athena/
Last Load Error:
/usr/local/frc/third-party/lib/libcscorejni.so: libwpinet.so: cannot open shared object file: No such file or directory
at edu.wpi.first.util.RuntimeLoader.loadLibrary(RuntimeLoader.java:94)
at edu.wpi.first.cscore.CameraServerJNI.<clinit>(CameraServerJNI.java:37)
at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:406)
at frc.robot.Main.main(Main.java:23)
java.io.IOException: cscorejni could not be loaded from path or an embedded resource.
attempted to load for platform /linux/athena/
Last Load Error:
/usr/local/frc/third-party/lib/libcscorejni.so: libwpinet.so: cannot open shared object file: No such file or directory
at edu.wpi.first.util.RuntimeLoader.loadLibrary(RuntimeLoader.java:94)
at edu.wpi.first.cscore.CameraServerJNI.<clinit>(CameraServerJNI.java:37)
at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:406)
at frc.robot.Main.main(Main.java:23)
java.io.IOException: cscorejni could not be loaded from path or an embedded resource.
attempted to load for platform /linux/athena/
Last Load Error:
/usr/local/frc/third-party/lib/libcscorejni.so: libwpinet.so: cannot open shared object file: No such file or directory
at edu.wpi.first.util.RuntimeLoader.loadLibrary(RuntimeLoader.java:94)
at edu.wpi.first.cscore.CameraServerJNI.<clinit>(CameraServerJNI.java:37)
at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:406)
at frc.robot.Main.main(Main.java:23)
java.io.IOException: cscorejni could not be loaded from path or an embedded resource.
attempted to load for platform /linux/athena/
Last Load Error:
/usr/local/frc/third-party/lib/libcscorejni.so: libwpinet.so: cannot open shared object file: No such file or directory
at edu.wpi.first.util.RuntimeLoader.loadLibrary(RuntimeLoader.java:94)
at edu.wpi.first.cscore.CameraServerJNI.<clinit>(CameraServerJNI.java:37)
at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:406)
at frc.robot.Main.main(Main.java:23)
java.io.IOException: cscorejni could not be loaded from path or an embedded resource.
attempted to load for platform /linux/athena/
Last Load Error:
/usr/local/frc/third-party/lib/libcscorejni.so: libwpinet.so: cannot open shared object file: No such file or directory
at edu.wpi.first.util.RuntimeLoader.loadLibrary(RuntimeLoader.java:94)
at edu.wpi.first.cscore.CameraServerJNI.<clinit>(CameraServerJNI.java:37)
at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:406)
at frc.robot.Main.main(Main.java:23)
java.io.IOException: cscorejni could not be loaded from path or an embedded resource.
[... 127 lines cut ...]
at frc.robot.Main.main(Main.java:23)
java.io.IOException: cscorejni could not be loaded from path or an embedded resource.
attempted to load for platform /linux/athena/
Last Load Error:
/usr/local/frc/third-party/lib/libcscorejni.so: libwpinet.so: cannot open shared object file: No such file or directory
at edu.wpi.first.util.RuntimeLoader.loadLibrary(RuntimeLoader.java:94)
at edu.wpi.first.cscore.CameraServerJNI.<clinit>(CameraServerJNI.java:37)
at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:406)
at frc.robot.Main.main(Main.java:23)
java.io.IOException: cscorejni could not be loaded from path or an embedded resource.
attempted to load for platform /linux/athena/
Last Load Error:
/usr/local/frc/third-party/lib/libcscorejni.so: libwpinet.so: cannot open shared object file: No such file or directory
at edu.wpi.first.util.RuntimeLoader.loadLibrary(RuntimeLoader.java:94)
at edu.wpi.first.cscore.CameraServerJNI.<clinit>(CameraServerJNI.java:37)
at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:406)
at frc.robot.Main.main(Main.java:23)
java.io.IOException: cscorejni could not be loaded from path or an embedded resource.
attempted to load for platform /linux/athena/
Last Load Error:
/usr/local/frc/third-party/lib/libcscorejni.so: libwpinet.so: cannot open shared object file: No such file or directory
at edu.wpi.first.util.RuntimeLoader.loadLibrary(RuntimeLoader.java:94)
at edu.wpi.first.cscore.CameraServerJNI.<clinit>(CameraServerJNI.java:37)
at edu.wpi.first.wpilibj.RobotBase.startRobot(RobotBase.java:406)
at frc.robot.Main.main(Main.java:23)

PeterJohnson, 2022-05-23:
We recently split the wpinet library out, and GR hasn’t yet been updated to match.  We will need to do a 2023 alpha 1 release before development artifacts are easily usable again.

Sheraz4201, 2022-05-23:
Is there any current ETA for this? We're interested in using the updated libraries in our offseason, specifically the fixes to the pose estimator.

PeterJohnson, 2022-05-23:
We'll talk about it later this week, so probably best case next weekend?

calcmogul, 2022-05-23:
You can copy the relevant class from https://github.com/wpilibsuite/allwpilib/tree/main/wpimath/src/main/java/edu/wpi/first/math/estimator in the meantime.

Starlight220, 2022-06-01:
I think 2023 alpha was released, so this can be closed?

sciencewhiz, 2022-06-01:
Gradlerio isn't published

PeterJohnson, 2022-06-01:
We had published it but needed to unpublish it due to some tools not being published.  I think those issues have been resolved so we can publish again.

PeterJohnson, 2022-06-02:
This should work now (GR 2023.0.0-alpha-1 has been released).

calcmogul, 2022-06-02:
It mostly worked. The deprecation removals broke REVLib.
```
/home/tav/.gradle/caches/transforms-3/156d6d5d888d571008412cf4792285e9/transformed/REVLib-cpp-2022.1.1-linuxathenastatic/linux/athena/static/libREVLib.a(CANSparkMaxLowLevel.o):(.data.rel.ro._ZTIN3frc15MotorControllerE[_ZTIN3frc15MotorControllerE]+0x8): undefined reference to `typeinfo for frc::SpeedController'
/home/tav/.gradle/caches/transforms-3/156d6d5d888d571008412cf4792285e9/transformed/REVLib-cpp-2022.1.1-linuxathenastatic/linux/athena/static/libREVLib.a(CANSparkMaxLowLevel.o):(.data.rel.ro+0x20): undefined reference to `frc::SpeedController::SetVoltage(units::unit_t<units::unit<std::ratio<1ll, 1ll>, units::base_unit<std::ratio<2ll, 1ll>, std::ratio<1ll, 1ll>, std::ratio<-3ll, 1ll>, std::ratio<0ll, 1ll>, std::ratio<-1ll, 1ll>, std::ratio<0ll, 1ll>, std::ratio<0ll, 1ll>, std::ratio<0ll, 1ll>, std::ratio<0ll, 1ll> >, std::ratio<0ll, 1ll>, std::ratio<0ll, 1ll> >, double, units::linear_scale>)'
```

calcmogul, 2022-06-03:
^ That's unrelated to this issue though. The alpha release fixed the linkage issue the OP mentioned.