wpilibsuite/allwpilib #4297: wpi/condition_variable.h: No such file or directory while building

bobbbay, 2022-06-05:
**Describe the question you have.**
I'm trying to build WPILib on my NixOS (Linux) machine. On `./gradlew halBaseLinuxathenaReleaseSharedLibrary --build-cache`, I get:

```
allwpilib on  HEAD (d66555e) via △ via ☕ v11.0.15 via ❄️  impure
❯ ./gradlew halBaseLinuxathenaReleaseSharedLibrary --build-cache -PtoolChainPath=/nix/store/qichdxfayyrzbir2hbpzjqxv08yig97w-devshell-dir/bin
Unable to list file systems to check whether they can be watched. Assuming all file systems can be watched. Reason: Could not query file systems: could not open mount file (errno 2: No such file or directory)
> Task :buildSrc:compileJava NO-SOURCE
> Task :buildSrc:compileGroovy UP-TO-DATE
> Task :buildSrc:processResources NO-SOURCE
> Task :buildSrc:classes UP-TO-DATE
> Task :buildSrc:jar UP-TO-DATE
> Task :buildSrc:assemble UP-TO-DATE
> Task :buildSrc:compileTestJava NO-SOURCE
> Task :buildSrc:compileTestGroovy NO-SOURCE
> Task :buildSrc:processTestResources NO-SOURCE
> Task :buildSrc:testClasses UP-TO-DATE
> Task :buildSrc:test NO-SOURCE
> Task :buildSrc:check UP-TO-DATE
> Task :buildSrc:build UP-TO-DATE

> Configure project :simulation:gz_msgs
Skipping gz msgs.

> Configure project :simulation:frc_gazebo_plugins
Skipping FRC Plugins.

> Configure project :simulation:halsim_gazebo
Skipping FRC Plugins.
Skipping builds for bionic (toolchain is marked optional)
Skipping builds for raspbian (toolchain is marked optional)

> Task :hal:generateUsageReporting UP-TO-DATE

> Task :hal:compileHalBaseLinuxathenaReleaseSharedLibraryHalBaseCpp FAILED
/home/bob/tmp/allwpilib/hal/src/main/native/cpp/Main.cpp:7:10: fatal error: wpi/condition_variable.h: No such file or directory
 #include <wpi/condition_variable.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

In file included from /home/bob/tmp/allwpilib/hal/src/main/native/cpp/handles/HandlesInternal.cpp:5:0:
/home/bob/tmp/allwpilib/hal/src/main/native/include/hal/handles/HandlesInternal.h:9:10: fatal error: wpi/Synchronization.h: No such file or directory
 #include <wpi/Synchronization.h>
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

/home/bob/tmp/allwpilib/hal/src/main/native/cpp/ErrorHandling.cpp:5:10: fatal error: fmt/format.h: No such file or directory
 #include <fmt/format.h>
          ^~~~~~~~~~~~~~
compilation terminated.

/home/bob/tmp/allwpilib/hal/src/main/native/cpp/cpp/fpga_clock.cpp:10:10: fatal error: fmt/format.h: No such file or directory
 #include <fmt/format.h>
          ^~~~~~~~~~~~~~
compilation terminated.

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':hal:compileHalBaseLinuxathenaReleaseSharedLibraryHalBaseCpp'.
> Multiple build operations failed.
      C++ compiler failed while compiling Main.cpp.
      C++ compiler failed while compiling HandlesInternal.cpp.
      C++ compiler failed while compiling ErrorHandling.cpp.
      C++ compiler failed while compiling fpga_clock.cpp.
  See the complete log at: file:///home/bob/tmp/allwpilib/hal/build/tmp/compileHalBaseLinuxathenaReleaseSharedLibraryHalBaseCpp/output.txt
   > C++ compiler failed while compiling Main.cpp.
   > C++ compiler failed while compiling HandlesInternal.cpp.
   > C++ compiler failed while compiling ErrorHandling.cpp.
   > C++ compiler failed while compiling fpga_clock.cpp.

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 22s
4 actionable tasks: 1 executed, 3 up-to-date
Unable to list file systems to check whether they can be watched. The whole state of the virtual file system has been discarded. Reason: Could not query file systems: could not open mount file (errno 2: No such file or directory)

Publishing build scan...
https://gradle.com/s/63jikv2vlfur2
```

**Additional context**
The toolchain is installed (`arm-...-gcc` works), but in a special location. I use `./gradlew ... -PtoolChainPath=/nix/store/qichdxfayyrzbir2hbpzjqxv08yig97w-devshell-dir/bin` but to no avail.

I'm running off of tag `v2022.4.1`.

calcmogul, 2022-06-05:
I was able to reproduce it on my Arch machine.
```bash
./gradlew halBaseLinuxathenaReleaseSharedLibrary
```
Seems like that rule is just broken.

ThadHouse, 2022-06-05:
The halBase tasks can not build a shared library. This is expected.

ThadHouse, 2022-06-05:
To add on to this, the *base tasks are there so shared and static binaries can link to the same set of object files (its a gradle hack). The static binary builds of the base task are what are used.

https://github.com/wpilibsuite/allwpilib/blob/main/shared/jni/setupBuild.gradle#L60

The above disables the shared build from running when `build` is ran, and then because of the return none of the dependencies are even added. All the buildable flag does is stop it from being a part of the build task. If the task is explicitly ran, as you're doing, it will force it to build, but because we never added any of the dependencies in shared library mode it fails.

bobbbay, 2022-06-05:
Thanks, that did it!

closed 2022-06-05 by bobbbay