wpilibsuite/allwpilib #4303: Migrate C++ to use `unique_ptr` semantics. This is mostly a change in documentation/recommended usage.

Starlight220, 2022-06-09:


PeterJohnson, 2022-06-21:
I'm not seeing how unique_ptr can be used for commands in a way that keeps the decorator approach.  The fundamental problem is the decorator functions on the class don't have access to the unique_ptr that is controlling the lifetime of the class.  So there’s no way for a decorator to return the same unique_ptr that currently owns this.  Using shared_ptr could work (with shared_from_this).

Starlight220, 2022-06-25:
After discussion, we're changing some details:
Instead of using `unique_ptr` directly, which would require *unintuitive* manipulation of `std::move` so we can use decorators (which don't have access to the `unique_ptr` owning the command), we'll be defining the decorators on `CommandPtr` -- a wrapper around `unique_ptr` that can be used with value semantics (and unlike `unique_ptr`, we can define methods on it).

PR #4319 (2022-10-05): [cpp commands] Migrate to unique_ptr
closes #4303 

Once this is more advanced, check whether renaming `CommandPtr` to `Command` (removing the current `Command` class at the head of the inheritance hierarchy) would help reducing breaking changes and keep Java/C++ syntax similarity.
closed 2022-10-05 by PeterJohnson