wpilibsuite/allwpilib #4320: Update to the latest version of vscode

mendax1234, 2022-06-22:
**Describe the question you have.**
A clear and concise description of what you want clarification on.

**Describe the reason for your confusion.**
A clear and concise description of why the question requires clarification and what's confusing.
l wonder whether l can update the vscode which is integrated in WPILib to the latest version. If not , what's the difference between the vscode which l download from microsoft and install the WPILib extension myself with the WPILib's integrated vscode 

**Is your question related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Nope

**Additional context**
Add any other context or screenshots about the question here.

Daltz333, 2022-06-22:
You can update the vscode installed by WPILib through normal means. I believe there is an update icon on the bottom right.

Standalone installations are not supported supported.

mendax1234, 2022-06-22:
> You can update the vscode installed by WPILib through normal means. I believe there is an update icon on the bottom right.
> 
> Standalone installations are not supported supported.

l still can't find it. Can you take a screenshot? Thanks!

PeterJohnson, 2022-06-23:
We use the portable install (https://code.visualstudio.com/docs/editor/portable).  So you need to download the .zip and extract it over the WPILib installed version (`C:\Users\Public\wpilib\2022\vscode` on Windows).

mendax1234, 2022-06-23:
> So you need to download the .zip and extract it over the WPILib installed version (C:\Users\Public\wpilib\2022\vscode on Windows).

So l can download the .zip file of the latest version of vscode and extract it in to the `C:\Users\Public\wpilib\2022\vscode` ? And then l can update my vscode?

mendax1234, 2022-06-23:
I have updated it and l copy the /data to the new folder and it seems to work well. But, l suggest that this should be done automatically. Thanks!

ThadHouse, 2022-06-23:
We disable automatic updates, as we’ve had updates in the past that have broken the extension. We can’t have that happen during the season. Which is why we ship a specific tested version and disable updates.

mendax1234, 2022-06-23:
Got it, thanks!

mendax1234, 2022-06-23:
> We disable automatic updates, as we’ve had updates in the past that have broken the extension. We can’t have that happen during the season. Which is why we ship a specific tested version and disable updates.

But l still recommnad adding the update button because l think maybe some of the people (like me) use vscode to write the code and want to update to the latest version of vscode to use some functions like Github copilot. For those who only use vscode to do frc code work , they can choose not to upgrade the vscode. l think this will be a double-win solution. @ThadHouse @PeterJohnson

calcmogul, 2022-06-23:
The problem with that is people will upgrade blindly during the season and potentially break their installs.

mendax1234, 2022-06-23:
> The problem with that is people will upgrade blindly during the season and potentially break their installs.

Emmm, l suggest that when people want to upgrade, a window will pop up saying something like you have to take the risk of upgrading. We( WPILib Team) don't promise the stability. l think many other softwares have this mechanism that we ourselves should take the risk of upgrading. Besides , l also recommand that WPILib can further test the latest vscode version to decide that which version of vscode may be stable and recommnad users to upgrade. If the version is not the one tested stable by the WPILib team , when people upgrade , he will receive a windwos poped up as l mentioned above.

PeterJohnson, 2022-06-23:
I don't think we have the ability to customize vscode's upgrade behavior to that extent.  Stability also means less support burden (which includes both our team and CSAs).

mendax1234, 2022-06-23:
> I don't think we have the ability to customize vscode's upgrade behavior to that extent.

Can the WPILIB Extension detect the behavior to upgrade the vscode?

> Experience has demonstrated that if given the option people will ignore every warning you put up and do it anyway, then ask for help when it breaks.

All right. l concede that you are right.

TheTripleV, 2022-06-23:
> some of the people (like me) use vscode to write the code and want to update to the latest version of vscode to use some functions like Github copilot.

Wpilib vscode and regular vscode can coexist on your machine. It doesn't have to be one vscode to rule them all.

mendax1234, 2022-06-23:
Thanks for replying. 

> > some of the people (like me) use vscode to write the code and want to update to the latest version of vscode to use some functions like Github copilot.
> 
> Wpilib vscode and regular vscode can coexist on your machine. It doesn't have to be one VS to code them all.

Thanks for replying!

Starlight220, 2022-06-23:
if I remember correctly, the taskbar grouping is much more ingrained in the OS than is worth messing with.

closed 2022-06-24 by mendax1234