wpilibsuite/allwpilib #4327: No usable version of libssl was found

mrhappyma, 2022-07-02:
Hello!

I am running Ubuntu Ubuntu 22.04 LTS x86_64. I downloaded the latest release of WPILib and followed the [Linux instillation instructions](https://docs.wpilib.org/en/stable/docs/zero-to-robot/step-2/wpilib-setup.html#extracting-the-installer). However when running the final `./WPILibInstaller` I am greeted with the following:
```shell
No usable version of libssl was found
Aborted (core dumped)
```

This seems to be that a package is just missing but I was unable to find instillation instructions. Help would be greatly appreciated :)

ThadHouse, 2022-07-02:
Currently the 2022 installer doesn’t support 22.04, because the updated OpenSSL does not work with the version of .NET the installer uses. This will be fixed for 2023.

ThadHouse, 2022-07-02:
https://github.com/wpilibsuite/WPILibInstaller-Avalonia/issues/255

See here for some more information.

Starlight220, 2022-07-05:
Can this be closed as resolved?

prateekma, 2022-07-27:
Closing due to inactivity. Please reopen if you are still unable to resolve the issue @mrhappyma.

closed 2022-07-27 by prateekma