wpilibsuite/allwpilib #4329: A strange error of the project

mendax1234, 2022-07-05:
[image]

These errors weren't there several hours ago, but they suddenly appear now. Can anybody help me with it?

Starlight220, 2022-07-05:
Does the code build?
If not, it's likely an Intellisense issue. Try using the `Clear Java workspace` command and/or reopening VS Code.

sciencewhiz, 2022-07-05:
Since you previously discussed upgrading VS Code, Does this occur when you use the supported version?

mendax1234, 2022-07-06:
> Since you previously discussed upgrading VS Code, Does this occur when you use the supported version?

The error occurs when l use the supported version

mendax1234, 2022-07-06:
> Does the code build? If not, it's likely an Intellisense issue. Try using the `Clear Java workspace` command and/or reopening VS Code.

Thanks, l am going to try thi.

closed 2022-07-06 by mendax1234