wpilibsuite/allwpilib #4330: The problem with Vector3D library

mendax1234, 2022-07-06:
l want to know whether WPILIB plans to make a Vector3D library?

PeterJohnson, 2022-07-06:
Already implemented in #4175.  Is there something You’re looking for that’s not in that PR?

mendax1234, 2022-07-06:
> 3D classes were implemented in #4175. Is there something You’re looking for that’s not in that PR?

> We don't plan on adding a Vector3d class. We're planning on removing the Vector2d class in the drive folder eventually too. Depending on the operations you need, you can use Translation2d, Translation3d, Eigen::Vector (C++), or org.ejml.simple.SimpleMatrix (Java).

I just want to know how to convert size, yaw and pitch value to a vector3D (or tranlation3d, l don't know whether there is any difference with vector3D and translation3D)

mendax1234, 2022-07-06:
> No, they are new for 2023. But you'll get what you're looking for with `new Translation3d(distance, new Rotation3d(roll, pitch, yaw))`

Thanks, so what's the difference with transaltion3d and a 3 dimension vector?

PeterJohnson, 2022-07-06:
A Translation3d can be used to represent a vector or a point.

mendax1234, 2022-07-06:
> A Translation3d can be used to represent a vector or a point.

Then what about the rotation3d, what's the difference between rotation3d and translation3d?

auscompgeek, 2022-07-06:
A Rotation3d represents a direction in 3D space. A Translation3d has a magnitude.

prateekma, 2022-07-27:
Closing due to inactivity. Please reopen if you have more questions related to this @mendax1234.

closed 2022-07-27 by prateekma