wpilibsuite/allwpilib #4343: Add a deadband filter

agasser, 2022-07-29:
WPILib provides some great filters in the `math.filter` package. I suggest adding a deadband filter.

A deadband is very commonly used in FRC for joystick input since joysticks don't center exactly at zero. It's pretty easy to implement, but nearly every team needs to build it.

calcmogul, 2022-07-29:
We already have that implemented as a static function in MathUtil, although I don't think it's correct to call it a filter.

https://first.wpi.edu/wpilib/allwpilib/docs/release/java/edu/wpi/first/math/MathUtil.html#applyDeadband(double,double)

agasser, 2022-07-29:
Oh nice! I didn't know about this util function. Thanks.

closed 2022-07-30 by PeterJohnson

Starlight220, 2022-07-31:
> We already have that implemented as a static function in MathUtil, although I don't think it's correct to call it a filter.

@calcmogul 
Even though it isn't a filter, maybe it might be a good idea to mention it in the filter section of frc-docs? Because discoverability appears to be the problem here.

calcmogul, 2022-07-31:
Since the primary use case is applying a deadband to joystick axes, it should be included in docs related to joysticks. MathUtil.applyDeadband() isn't something that would get used in a measurement filter chain.