wpilibsuite/allwpilib #4355: Add conversions to and from vendor units to the Units utility class

calcmogul, 2022-08-16:
Teams often need to convert from SI units to the unorthodox units the vendors CTRE and REV use. We could add conversion functions to edu.wpi.first.math.util.Units in Java and add custom unit aliases to C++ to make that easier.

Users generally need to convert positions, velocities, feedforward gains, and feedback gains. Each vendor has different units for each of these.

The conversion names should include the vendor names to make it clear what devices they're for.

sciencewhiz, 2022-08-16:
Is there a way for vendors to add this? Specifically since CTRE has talked about changing the way they do units.

calcmogul, 2022-08-16:
Vendors could add conversion utilities to their own vendordeps, but there's no guarantee the APIs will be consistent between WPILib, CTRE, and REV.

calcmogul, 2023-08-01:
OBE now that vendor APIs generally use SI units.

closed 2023-08-01 by calcmogul