wpilibsuite/allwpilib #4404: Add joystick normalization for swerve

JoshuaAN, 2022-09-06:
The current implementation of desaturateWheelSpeeds allows joystick inputs to saturate for swerve drive.

For proper normalization, the following conditions must be met simultaneously with no discontinuous scaling applied:
- The joystick of maximum magnitude fully extends.
- The fastest swerve module reaches maximum velocity.

An implementation of the algorithm is shown in this post.
https://www.chiefdelphi.com/t/good-vs-bad-swerve/414439/32

calcmogul, 2022-09-06:
This is talking about SwerveDriveKinematics.desaturateWheelSpeeds(), to be clear.

calcmogul, 2022-09-06:
To avoid breakage, we can add a second overload of desaturateWheelSpeeds().

closed 2022-11-17 by PeterJohnson