wpilibsuite/allwpilib #4424: RobotBase Skeleton Java Template improve example loop flow

tom131313, 2022-09-18:
[RobotBase Skeleton Java Template](https://github.com/wpilibsuite/allwpilib/blob/main/wpilibjExamples/src/main/java/edu/wpi/first/wpilibj/templates/robotbaseskeleton/Robot.java#L38-L44)

The template Java code needs a small change to be much more easy to follow. It's helpful but essentially dysfunctional without fixing it.

The problem is replicated four times for the four modes so my comments **apply to all four code blocks**.
This runs `disabled()` only once then hangs in a loop waiting for data and Enabled:
```
     if (isDisabled()) {
        DriverStation.inDisabled(true);
        disabled();
        DriverStation.inDisabled(false);
        while (isDisabled()) {
          DriverStation.waitForData();
        }
```
A better example would be to move the `while`
```
     if (isDisabled()) {
        while (isDisabled()) {
         DriverStation.inDisabled(true);
         disabled();
         DriverStation.inDisabled(false);
         DriverStation.waitForData();
        }
```
Adding a couple more examples or comments would be helpful such as:
```
DriverStation.waitForData(); // () (0) (seconds)
DriverStation.isNewControlData();
```

ThadHouse, 2022-09-18:
That’s actually how the skeleton is designed. We generally don’t recommend it be used unless teams really want to avoid timed robot, or think that the old sample robot was good (it wasn’t). Timed robot works way better if you want a function to run periodically and not just once on init.

tom131313, 2022-09-18:
Obviously I don't know why the skeleton is designed that way and still don't so I thought my design was better since it doesn't fail as the existing skeleton irritatingly does (maybe that's the point of the design?).

I had a very special use that was slightly easier to implement in RobotBase. Probably CommandBase would have accomplished it almost easily.

ThadHouse, 2022-09-18:
The point of the design is to call the function once, and then the user writes loops in there, and also has to check if the state has changed. Thats how things worked in the very old days (Think like 2010 and before) and some people still desperately hold onto the belief that we need to provide a template like that. Its harder in almost every way. The old templates if you returned didn't repeat (This was mainly done for autonomous purposes, you could just write your auto once, and wouldn't have to add code to make it not repeat if it ended early).

tom131313, 2022-09-19:
Thanks. I correctly guessed the design after awhile. I'm a zealot for documenting intentions and assumptions. The documentation Note says the startCompetition is blank. Fortunately that isn't true. The design is good and I would have been happy if the documentation says the robot program loops forever running the method for the selected mode once then waiting for disable or another mode to be selected.

closed 2022-09-19 by tom131313