wpilibsuite/allwpilib #4454: robot pausing for about 2 seconds while switching between trajectories

korozya-bit, 2022-10-03:
while coding our autonoumas commands using the Trajectory class we found it easier to use a couple of diffrent Trajectories, rather the one long trajectory with a bunch of waypoints.
When we ran the auto we noticed that the robot pauses for about 2 seconds every time it switched from one Trajectory to another.
 
Steps to reproduce the behavior:
1. Run two trajectories sequentially

**Expected behavior**
The robot will drive through the first auto, pause, and then drive through the second auto

 - WPILib Version: 2022.4.1
 - OS: Windows: 10
 - Java version: 11.0.13

Starlight220, 2022-10-03:
Trajectories by default have a start and end velocity of 0, so maybe that's the pause you're seeing?

korozya-bit, 2022-10-03:
1. is it something that can be changed or is it hardcoded?
And if so then how do you change it?

2. it kind of wait's a little before accelerating again it doesn't immediately start going

Starlight220, 2022-10-03:
There are `startVelocity` and `endVelocity` fields in the `TrajectoryConfig` struct that can be changed.

korozya-bit, 2022-10-04:
ok thanks, thats much better

closed 2022-10-04 by korozya-bit