wpilibsuite/allwpilib #4455: Install on Fedora linux

KilroyWasHere-cs-j, 2022-10-04:
Attempting to install on Fedora Linux returns  
No usable version of libssl was found
Aborted (core dumped)
installing openssl-devel does not fix the problem

PeterJohnson, 2022-10-04:
The 2022 installer requires an older version (1.x) of OpenSSL.  The 2023 installer will support newer versions.  Unfortunately there is no workaround for 2022 other than figuring out a way to install an older OpenSSL version.

smoser, 2022-10-07:
I hit this installing on Ubuntu 22.04.
I worked around by building a openssl 1.1, putting it into LD_LIBRARY_PATH and running the installer.  It looks like no other issues.  So, you could probably get by with grabbing an Fedora RPM (older release) and running.

    LD_LIBRARY_PATH=$PWD/usr/lib/x86_64-linux-gnu/ ./WPILib_Linux-2022.4.1/WPILibInstaller

That seemed to work.
And hopefully the first wpilib for 2023 is due soon enough that it wont be a problem.

JonathanLindsey, 2022-11-28:
I had the same problem on Pop!_OS (Ubuntu based). Instead of messing around with openssl, I opted to install the regular VS Code and then the FRC extensions along with a Java JDK. I had a little issue with (simulating) the CANSparkMax and had to do the offline install method but that's probably a completely different issue. Disclaimer: I have not tried deploying to a roboRIO yet. Your millage may vary. The official install method may very well have other advantages such as being able to be used with years old projects and may be more stable (better for competition).

Depending on your distribution it should be something like this.

Using the apt package manager install the Java JDK either with the `default-jdk` or `openjdk-11-jdk` which `default-jdk` currently points to.
```
sudo apt install default-jdk
# or: sudo apt install openjdk-11-jdk
```

Install VS Code with either the package from your distribution's repositories or as a snap package.
```
sudo apt install code
```
If that gives an error, your distribution's repositories doesn't have VS Code (Ubuntu). Install snapd if you don't already have it and then VS Code as a snap.
```
sudo apt install snapd
sudo snap install code --classic
```

Install the VS Code extensions for FRC. This can be done manually in VS Code or with this command.
```
code --install-extension wpilibsuite.vscode-wpilib --install-extension redhat.java --install-extension vscjava.vscode-java-dependency --install-extension vscjava.vscode-java-debug
```

I also like to disable VS Code's telemetry: `F1` or `Ctrl + Shift + P`, type "Settings UI", select the appropriate list item and search for "telemetry".

I haven't tried C++ but I would image that would take just a few more steps.

ThadHouse, 2022-11-28:
One thing to note of the above, is we don’t always update the extension in the marketplace, and in the future we plan on making the one on the marketplace just point to the installer. You can install the vsix manually, but like always we don’t recommend this. We do know there are some breakages with newer versions of vscode And the 2022 extensions (the project importer randomly works or doesn’t). 

The main advantages of the installer is it’s a complete setup that is known to work. Randomly installing all the different pieces from multiple places has a higher probability of things going wrong. Such as broken jdks, vscode not properly picking up the right jdk, the offline dependency cache being deleted during competition, completely breaking builds without internet, and many other things. The first debugging step we will always do is ensure you’re using the installer, and if not we can’t guarantee things will work or we can take the time to debug. 

This years been hard with both the vscode git changes and OpenSSL 3, but both of those issues have already been fixed for 2023.

JonathanLindsey, 2023-01-21:
The 2023 installer worked for me on Pop_OS 22.04 LTS (based on Ubuntu 22.04 LTS) so this issue can probably be closed.

closed 2023-01-21 by auscompgeek