wpilibsuite/allwpilib #4468: SequentialCommandGroup crashes with IndexOutOfBoundsException with isFinished override

Nyxyxylyth, 2022-10-12:
**Describe the bug**
SequentialCommandGroup crashes with IndexOutOfBoundsException if isFinished is overridden to return false

**To Reproduce**
Steps to reproduce the behavior:
1. See minimal example at https://github.com/FRC-1481-Riveters/CommandCrash.git
2. in RobotContainer.java:
    new JoystickButton( m_controller, 8 )
    .whileActiveOnce( new ExampleCommand() );
3. in ExampleCommand.java:
public class ExampleCommand extends SequentialCommandGroup {
  public ExampleCommand() {
    addCommands(
      new WaitCommand( 1.0 )
    );
  }
  // Returns true when the command should end.
  @Override
  public boolean isFinished() {
    return false;
  }
4. Hold down start button on robot or in PC Sim for _longer_ than one second, and WPILib crashes on line 65 of SequentialCommandGroup.class, because the index points one _past_ the array, and isFinished doesn't get to terminate as usual:
    Command currentCommand = m_commands.get(**m_currentCommandIndex**);

**Expected behavior**
Command should just keep executing unless stopped by some other mechanism

**Desktop (please complete the following information):**
 - WPILib Version: 2022.4.1
 - OS: Windows 11
 - Java version 11.0.13+8

**Additional context**
Why doesn't execute() have the same bounds checking that end() has?  If it just checked m_currentCommandIndex before using it, it would continue executing.  Granted, the command wouldn't terminate via isFinished, but it could still be canceled externally, or stop when the button is released.
I get that overriding isFinished() is not a great idea here, but it'd be more graceful (and obvious) if the command simply kept running, instead of having to debug the guts of SequentialCommandGroup to understand the exception.
Simply modifying the isEmpty() check in execute() would probably do the trick:
    if (m_commands.isEmpty() || (m_currentCommandIndex == m_commands.size)_ ) {

Starlight220, 2022-10-12:
Overriding lifecycle methods when inheriting from commands other than `Command`/`CommandBase` is strongly discouraged and not really supported.
A PR preventing overriding those methods on command groups was already merged.

closed 2022-10-12 by Starlight220

Nyxyxylyth, 2022-10-12:
Splendid, thanks!  I better get on the 2023 alpha train.