wpilibsuite/allwpilib #4484: Deployment is Seccusful but there is no robot code, and no errors in dashboard.

GlueGunGlitter, 2022-10-19:
When I try to deploy my code to the robot, the terminal shows that the build is successful, but the dashboard displays no robot code. also, phoenix tuner shows that ther is no CAN devices connected, although the PDP is fully conected.

Terminal output from VS code:
`> Task :discoverroborio
Discovering Target roborio
Using admin@10.67.40.2:22 for target roborio
Driver Station reported IP: 10.67.40.2

> Task :deployprogramKillroborioroborio
    -C-> sed -i -e 's/"exec /"/' /usr/local/frc/bin/frcRunRobot.sh @ /home/lvuser
      -[-1]
    -C-> . /etc/profile.d/natinst-path.sh; /usr/local/frc/bin/frcKillRobot.sh -t 2> /dev/null @ /home/lvuser
    -C-> sed -i -e 's/^StartupDLLs/;StartupDLLs/' /etc/natinst/share/ni-rt.ini @ /home/lvuser
      -[-1]

> Task :deployjrefrcJavaroborio
    -C-> if [[ -f "/usr/local/frc/JRE/bin/java" ]]; then echo OK; else echo MISSING; fi @ /tmp
      -[-1]-> OK

    -C-> opkg list-installed | grep openjdk @ /tmp
      -[0]-> frc2021-openjdk-11-jre - 11.0.9u11-1 - FRC OpenJDK Java Runtime Environment

  Searching for JRE 11.0.9u11-1
  Found JRE frc2021-openjdk-11-jre - 11.0.9u11-1 - FRC OpenJDK Java Runtime Environment
  JRE Is Correct Version
  Artifact skipped

> Task :deployfrcStaticFileDeployroborio
    -C-> mkdir -p  @ /home/lvuser/deploy
      -[-1]
    1 file(s) are up-to-date and were not deployed

> Task :deployrobotCommandfrcJavaroborio
    -C-> echo '/usr/local/frc/JRE/bin/java -XX:+UseConcMarkSweepGC -XX:+AlwaysPreTouch -Djava.lang.invoke.stringConcat=BC_SB -Djava.library.path=/usr/local/frc/third-party/lib -jar "/home/lvuser/Robot-2023.jar" ' > /home/lvuser/robotCommand @ /home/lvuser
      -[-1]
    -C-> chmod +x /home/lvuser/robotCommand; chown lvuser /home/lvuser/robotCommand @ /home/lvuser

> Task :deployfrcJavaroborio
    1 file(s) are up-to-date and were not deployed
    -C-> chmod +x "/home/lvuser/Robot-2023.jar"; chown lvuser "/home/lvuser/Robot-2023.jar" @ /home/lvuser

> Task :deploynativeZipsfrcJavaroborio
    26 file(s) are up-to-date and were not deployed
    -C-> chmod -R 777 "/usr/local/frc/third-party/lib" || true; chown -R lvuser:ni "/usr/local/frc/third-party/lib" @ /usr/local/frc/third-party/lib
    -C-> ldconfig @ /usr/local/frc/third-party/lib

> Task :deployprogramStartfrcJavaroborio
    -C-> sync @ /home/lvuser
      -[-1]
    -C-> . /etc/profile.d/natinst-path.sh; /usr/local/frc/bin/frcKillRobot.sh -t -r 2> /dev/null @ /home/lvuser

BUILD SUCCESSFUL in 4s
13 actionable tasks: 11 executed, 2 up-to-date
 *  Terminal will be reused by tasks, press any key to close it. `

Dashboard log:
`﻿﻿﻿﻿﻿﻿﻿﻿Warning ﻿﻿ 44000 ﻿﻿ DS Disable ﻿﻿ Driver Station ﻿﻿﻿
﻿﻿﻿﻿﻿﻿﻿﻿Warning ﻿﻿ 44000 ﻿﻿ DS Disable ﻿﻿ Driver Station ﻿﻿﻿
﻿﻿﻿﻿﻿﻿﻿﻿Warning ﻿﻿ 44000 ﻿﻿ DS Disable ﻿﻿ Driver Station ﻿﻿﻿
`

I can't seem to understand from where the problem is coming from, but I think it's from the Roborio firmware or some WPI installation. I'm not the only one that has experienced this problem. When I searched I found [this redit post](posthttps://www.reddit.com/r/FRC_PROGRAMMING/comments/f7yz3t/no_robot_code_when_deploy_successful/), and [this delphi post] (https://www.chiefdelphi.com/t/code-deploys-successfully-in-ide-but-shows-no-robot-code-error-on-driver-station/401844). The guy from the delphi post said that he solved his problem through replacing the PDP and reimaging the Roborio, but reimaging didn't work for me. Although the problems origin is probably some wrong installation, the fact that there is no error is misleading. 

 - WPILib Version: 2022.4.1
 - OS: Windows 10
 - Java version 11.0.9u11-1
 - C++ version 7.3.0

prateekma, 2022-10-19:
Is that the only output in the Driver Station log?

sciencewhiz, 2022-10-19:
Is there any output in riolog?

Starlight220, 2022-10-19:
I don't know if this is the right project, but in https://github.com/GlueGunGlitter/Robot-2023/blob/main/build.gradle#L3 you're using WPILib 2022.2.1. Since this didn't error on deploy, I'm assuming you're using an old RIO image. I'm not sure this would cause the issue here, but it might be relevant.
Also, make sure you're not somehow mismatching RIO1/RIO2 images.

PeterJohnson, 2022-10-19:
If reimaging didn't fix it, I've occasionally seen this happen if there's an infinite recursion during program startup, as the error message can be too big for it to be sent successfully to the DS.  Linking to your code would help.

prateekma, 2022-10-19:
You can also try deploying one of the template projects in VS Code without any modifications. If the issue is resolved, it must be something with your code as Peter mentioned above.

Starlight220, 2022-11-27:
Fixed by wpilibsuite/gradlerio#618?

PeterJohnson, 2022-11-28:
Yes, that should fix it.

closed 2022-11-28 by PeterJohnson