wpilibsuite/allwpilib #4513: Swerve Drive Pose Estimator example does not use pose heading for field-relative.

shueja-personal, 2022-10-24:
https://github.com/wpilibsuite/allwpilib/blob/023a5989f8bafec69c3650aca868c8b0cc4a6f3b/wpilibjExamples/src/main/java/edu/wpi/first/wpilibj/examples/swervedriveposeestimator/Drivetrain.java#L65

MrRedness, 2022-11-09:
Is it supposed to use the pose heading? I think it makes more sense relying on the gyro

calcmogul, 2022-11-09:
It should use the pose estimate's heading because that includes corrections from computer vision (not that it matters right now because bias in the gyro measurement will pull it off-course again).

MrRedness, 2022-11-09:
In my experience, gyro has always been more accurate than odometry or vision, at least on a navx2.

-------- Original Message --------

calcmogul, 2022-11-09:
Depends on how well your vision pipeline is tuned. Retroreflective tape sucks at pose estimation without a lot of work and a high resolution camera.

If your gyroscope really is more accurate, you should be setting the measurement standard deviation low enough relative to your vision that it's the one primarily driving the heading estimate. In that case, you'd still use the pose estimate over the raw measurements.

ElliotScher, 2023-11-07:
Should this be changed in both c++ and java examples?

PR #5900 (2023-11-10): replaced gyro rotation with poseEstimator rotation
closes #4513
closed 2023-11-10 by PeterJohnson