wpilibsuite/allwpilib #4535: Transform3D transformBy not behaving as I would expect

dellagd, 2022-10-31:
First time doing a lot with 3D vectors, so very possible I am just misunderstanding how this would work. However, it appears to me that the Pose3d's transformBy function is not computing the new Rotation3d for the resulting Pose3d correctly.

When using transformBy, I would expect the Transform3d's rotation to be applied starting at the Pose3d's current orientation. In other words, the transform is applied as if you place yourself in the orientation of the Pose3d, and do all the rotations using that orientation's axes, not the axis of the global frame. It doesn't look like that's happening.

**To Reproduce**
`pose.transformBy(transform)` returns a new pose that is `pose.getRotation().plus(transform.getRotation())`. If we were to do two transforms sequentially, we would have `pose.getRotation().plus(transform1.getRotation()).plus(transform2.getRotation())`. In the example below, I simplify this using just Rotation3ds to be `p.plus(t1).plus(t2)`

    Rotation3d r = new Rotation3d(
      Units.degreesToRadians(90.0),
      Units.degreesToRadians(45.0),
      Units.degreesToRadians(0.0)
    );

    var a = new Rotation3d(
      Units.degreesToRadians(-90.0),
      Units.degreesToRadians(0.0),
      Units.degreesToRadians(0.0)
    );

    var b = new Rotation3d(
      Units.degreesToRadians(-0.0),
      Units.degreesToRadians(-45.0),
      Units.degreesToRadians(0.0)
    );

These three rotations, transforming an object sequentially from a null rotation, should place you back into a null rotation. However, they do not, yielding roll-pitch-yaw of `[   35.26,   -30.00,   -54.74]`. However, if we do the rotations "backwards" with the `plus` method, we do get back to a null rotation of `[    0.00,     0.00,     0.00]`

    var t1 = b.plus(a).plus(r); //[ 35.26, -30.00, -54.74]
    var t2 = r.plus(a).plus(b); //[ 0.00,  0.00,  0.00]

`t1` shows what happens currently in `transformBy`. t2 is what happens if `transformBy` reverses the order of the rotation when it constructs the rotation component of the pose it returns, and what I think is what should be happening.

I don't know if this issue is just with how `transformBy` is setup, or if it is fundamental to the `plus` operation on Rotation3d.

Could be totally misusing these functions though! If so I would kindly appreciate some help in using `transformBy` correctly. Thanks!

calcmogul, 2022-10-31:
Rotations are always extrinsic (using the global coordinate frame). If you want intrinsic rotations, you have to apply the rotations in the opposite order.

dellagd, 2022-10-31:
Is this opposite order what should happen in `transformBy`? I am attempting to use `transformBy` to transform the position of a Pose3d for object A (global coordinate frame) using a Transform3d that describes how you move from A to B in the coordinate frame of object A.

calcmogul, 2022-10-31:
This is how Pose3d applies a transformation:
```java
  public Pose3d transformBy(Transform3d other) {
    return new Pose3d(
        m_translation.plus(other.getTranslation().rotateBy(m_rotation)),
        m_rotation.plus(other.getRotation()));
  }
```

calcmogul, 2022-10-31:
I think what you're trying to do is what this function does:
https://github.com/wpilibsuite/allwpilib/blob/main/wpimath/src/main/java/edu/wpi/first/math/ComputerVisionUtil.java#L137-L149

dellagd, 2022-10-31:
Yes, sorry if I wasn't clear, my original issue is questioning the rotation component of that very function. I am questioning if the function should be as follows in order to properly create a Pose3d that describes how the original Pose3d has been changed after applying the transform.
```
  public Pose3d transformBy(Transform3d other) {
    return new Pose3d(
        m_translation.plus(other.getTranslation().rotateBy(m_rotation)),
        other.getRotation().plus(m_rotation));
  }
```

dellagd, 2022-10-31:
Yeah, basically trying to do that: https://github.com/wpilibsuite/allwpilib/blob/main/wpimath/src/main/java/edu/wpi/first/math/ComputerVisionUtil.java#L137-L149

However, when I was testing this, I had to redo the `transformBy` function as I suggested above to get results that looked correct.

calcmogul, 2022-10-31:
If I change the TransformBy() implementation to the code you suggested, all the tests still pass, so you're probably right and there's gaps in our tests.

Could you add a test case to https://github.com/wpilibsuite/allwpilib/blob/main/wpimath/src/test/java/edu/wpi/first/math/geometry/Transform3dTest.java that can distinguish between the two implementations and prove one of them is correct?

dellagd, 2022-10-31:
I will give that a shot, thanks.

calcmogul, 2022-10-31:
> What exact proof would you look for?

The new test should pass for one version of the code and fail for the other so that the implementation is properly constrained to do the correct thing.

> Does my original example make sense?

Honestly, I don't understand what you were trying to say in your original post. Maybe it's too much for my brain on a Sunday night after a long weekend.

dellagd, 2022-11-02:
Added test case in https://github.com/wpilibsuite/allwpilib/pull/4545
Failure can be observed in the CI run. This test case succeeds when reversing the order of the rotations: https://github.com/wpilibsuite/allwpilib/actions/runs/3374965920/jobs/5601151619

closed 2022-11-07 by PeterJohnson