wpilibsuite/allwpilib #4550: Remove the state, input, and output generics from SwerveDrivePoseEstimator.java

calcmogul, 2022-11-02:
The C++ version takes the number of modules, then computes the number of states, inputs, and outputs internally. If we use `?` for some of the member generics like we've done in UnscentedKalmanFilter, we should be able to defer the generic type resolution so the user doesn't have to do it.

jlmcmchl, 2022-11-02:
Probably belongs in this issue - the alternative Y vector for correction should be documented to include module positions as well.

https://github.com/wpilibsuite/allwpilib/blob/main/wpimath/src/main/java/edu/wpi/first/math/estimator/SwerveDrivePoseEstimator.java#L53
https://github.com/wpilibsuite/allwpilib/blob/main/wpimath/src/main/native/include/frc/estimator/SwerveDrivePoseEstimator.h#L50

calcmogul, 2022-11-02:
That should be a separate issue since its fix is unrelated to what would fix this one.

closed 2022-12-02 by PeterJohnson