wpilibsuite/allwpilib #4568: 2023 beta outlineviewer crashes when connecting to networktables at localhost with NT4

fovea1959, 2022-11-04:
**Describe the bug**
If I trying to connect as a NT4 client to the networktables running in a simulation on the same PC (hostname localhost), outlineviewer simply disappears.

**To Reproduce**
Steps to reproduce the behavior:
1. Load up WPILIBJ and VScode from WPILib_Windows-2023.1.1-beta-2.iso.
2. Go into VSCODE 2023, create and simulate a Java command based template project. No need to add anything to the generated template.
3. Start outlineviewer. Select Options | Settings, Mode = Client NT4,  Team/IP localhost, Network Identify leave as "outlineviewer", uncheck "Get address from DS", hit apply button.
4. watch outlineviewer disappear.

**Expected behavior**
Outlineviewer should display the network tables from the running simulation. Outlineview does this correctly if running as an NT3 client.

**Screenshots**
(None)

**Desktop (please complete the following information):**
 - WPILib Version: 2023.1.1-beta2
 - OS: Windows 10
 - Java version: (bundled with WPILib 2023.1.1-beta2
 - C++ version: N/A

**Additional context**
None

ThadHouse, 2022-11-05:
We're having trouble reproducing this. Would you be able to try a few debug steps.

1. Install WinDbg Preview from the windows store and open it.
2. Go To File -> Launch Executable
3. Go to the outlineviewer executable (Its in C:\Users\Public\wpilib\2023\tools) and click open on it.
4. Its going to stop on exit immediately. Hit "Go"
5. Reproduce the error in the gui. This should print out the exception, and a stack trace of where the error occurs. If it does enter a breakpoint, but doesn't print the stacktrace to the command window, enter `k` in the command window input, and that will print out a stack trace.

fovea1959, 2022-11-06:
I'll have to close the ticket for now; I can't get it to happen today either. <sigh>

closed 2022-11-06 by fovea1959