wpilibsuite/allwpilib #4574: Edge-based BooleanEvents can't be used in multiple places

Starlight220, 2022-11-06:


Oblarg, 2022-11-12:
We probably need to do some cloning to make this work.

KangarooKoala, 2023-07-09:
Is this still an issue? Notably, multiple bindings on a `Trigger` assumes that calls to `getAsBoolean()` within a single poll of an `EventLoop` will all return the same value, so this should either be expanded to include fixing `Trigger` as well or ignored as "reusing BooleanSuppliers that behave this way isn't supported".

Starlight220, 2023-07-09:
Trigger's implementation was reverted back to not use BooleanEvent for this reason, so currently nothing uses BooleanEvent.
The difference that causes Trigger not to be affected by this is that Trigger has the edge logic in the action (and as such, isn't shared between bindings), whereas BooleanEvent has that logic in the condition (which is shared between bindings, hence the problem).

But yes, if Trigger would be given a non-idempotent condition (such as a count-based filter) it could cause unintentional behavior. We can avoid passing non-idempotent conditions in library code (hence, this issue), but there's not much we can do about user code.

KangarooKoala, 2023-07-10:
So, should we still fix `BooleanEvent`, and should we allow `Trigger` to handle non-idempotent conditions or not allow them?

Starlight220, 2023-07-10:
BooleanEvent should be fixed.

PR #4673 (2022-11-28): [cmd] Revert Trigger implementation
This might fix #4574 and similar.

There are no user-facing changes in this PR.
PR #5436 (2023-10-05): Make BooleanEvent more consistent
Each `BooleanEvent` now contains an internal state variable that is changed by an action bound to the event loop. When the loop is polled, the state variable changes to match the condition state. The lambda in `ifHigh` references the state variable instead of directly getting the condition state, meaning if multiple actions are bound to a `BooleanEvent`, they will all run if the condition is true at loop poll, even if the condition state changes mid loop.
Also, C++ had no test for `BooleanEvent`, so I copied the Java test and made a test.
Fixes #4574
closed 2023-10-05 by PeterJohnson