wpilibsuite/allwpilib #4602: Remove wheel speeds structs from pose estimators

calcmogul, 2022-11-10:
The wheel positions are doing most of the work in the pose estimator filters, so the wheel speeds are redundant. The duplication has confused several beta teams.

MrRedness, 2022-11-15:
Could m_observer.Predict() simply be removed, and only .Correct used?

calcmogul, 2022-11-15:
Predict() has side effects (updating the error covariance), so no.

MrRedness, 2022-11-15:
Knew it couldn't be that easy. I probably won't have time to dig into the linear algebra of this anytime soon, unfortunately.

calcmogul, 2022-11-15:
We're actually planning on rewriting it to not use a UKF.

* Use odometry internally for prediction
* Compute steady-state Kalman gain internally in ctor via KalmanFilter class
* Do vision correction via the following, using Twist2d for pose interpolation:
```
x = x + K * measuredPose.RelativeTo(predictedPose)
x = x + K(measured - x)
x = x + K measured - Kx
x = (1 - K)x + K measured
```

MrRedness, 2022-11-15:
That looks great! What does K represent in your example?

calcmogul, 2022-11-15:
The return value from KalmanFilter(...).getK().

MrRedness, 2022-11-15:
Makes sense 👍 
Thanks

closed 2022-12-02 by PeterJohnson