wpilibsuite/allwpilib #4603: Command factories and decorators don't return Sendable

Starlight220, 2022-11-10:
The factories and decorators return `Command`, not `CommandBase`, so they can't be put to the dashboard. An unsafe cast should work, given that there aren't any command classes that don't inherit `CommandBase`, but we should find a better (and safer) solution.

virtuald, 2022-11-10:
Commands are all polymorphic, so you can use dynamic cast instead?

Oblarg, 2022-11-12:
At this point, we may be better absorbing the `Command` and `Subsystem` interfaces into the base classes (or vice-versa).  I can't really think of many situations where users would actually want to implement their own `Command` from scratch without using the base class.

Starlight220, 2022-11-12:
At a second look, overriding `withName` in `CommandBase` should be enough for Java decorators, and we'll need to make sure we return `CommandBase` in the factories.
As for C++, that would be a different story: idk if CommandPtr usages are likely to be put to dashboard; if they are we might need to change `get()` to return `CommandBase*` instead of `Command*`. Non-CommandPtr usages would be stored by value as CommandBase subtypes anyway.

---
> At this point, we may be better absorbing the Command and Subsystem interfaces into the base classes (or vice-versa). I can't really think of many situations where users would actually want to implement their own Command from scratch without using the base class.

This is even more extreme in C++, with CommandHelper.
I'm not opposed, but that's _majorly_ breaking and should be discussed more in detail with a rollout schedule.

PR #4655 (2022-11-17): [cmd] Change factories return type to CommandBase
Fix #4603
closed 2022-11-17 by PeterJohnson