wpilibsuite/allwpilib #4625: DataLogManager singleton needs a Stop function

virtuald, 2022-11-13:
**Is your feature request related to a problem? Please describe.**
RobotPy tests crash on exit when DataLogManager was started. There isn't a way to stop it.

```
* thread #4, stop reason = signal SIGABRT
  * frame #0: 0x00007ff80ad9700e libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007ff80adcd1ff libsystem_pthread.dylib`pthread_kill + 263
    frame #2: 0x00007ff80ad18d24 libsystem_c.dylib`abort + 123
    frame #3: 0x00007ff80ad89082 libc++abi.dylib`abort_message + 241
    frame #4: 0x00007ff80ad7a1a5 libc++abi.dylib`demangling_terminate_handler() + 242
    frame #5: 0x00007ff80ac76e19 libobjc.A.dylib`_objc_terminate() + 104
    frame #6: 0x00007ff80ad884a7 libc++abi.dylib`std::__terminate(void (*)()) + 8
    frame #7: 0x00007ff80ad8ad05 libc++abi.dylib`__cxxabiv1::failed_throw(__cxxabiv1::__cxa_exception*) + 27
    frame #8: 0x00007ff80ad8accc libc++abi.dylib`__cxa_throw + 116
    frame #9: 0x00007ff80ad3595d libc++.1.dylib`std::__1::__throw_system_error(int, char const*) + 77
    frame #10: 0x00007ff80ad2d72d libc++.1.dylib`std::__1::mutex::lock() + 29
    frame #11: 0x00000001075d8fa1 libwpilibc.dylib`wpi::EventVector::Remove(unsigned int) + 17
    frame #12: 0x0000000107623647 libwpilibc.dylib`(anonymous namespace)::Thread::Main() + 3367
    frame #13: 0x0000000104f170fe libwpiutil.dylib`void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, wpi::detail::SafeThreadOwnerBase::Start(std::__1::shared_ptr<wpi::SafeThreadBase>)::$_0> >(void*) + 46
    frame #14: 0x00007ff80adcd4e1 libsystem_pthread.dylib`_pthread_start + 125
    frame #15: 0x00007ff80adc8f6b libsystem_pthread.dylib`thread_start + 15
```

**Describe the solution you'd like**

Either provide a public Stop function or something in frc::impl. Or make SafeThread safer somehow.

PeterJohnson, 2022-11-15:
This is actually caused by a cross-module call to DriverStation.  C++ has no guarantees on destruction order between modules.  In a few other places we check an atomic that's set during destruction of static instances so we should do the same here within DriverStation.

gaborszita, 2023-02-21:
Can you give me an example of where you "check an atomic that's set during destruction of static instances"? I may try to fix this myself if it isn't very complex.

gaborszita, 2023-02-21:
@PeterJohnson Can you give me an example of where you "check an atomic that's set during destruction of static instances"? I may try to fix this myself if it isn't very complex.

PeterJohnson, 2023-02-21:
One place we do it: https://github.com/wpilibsuite/allwpilib/blob/main/wpiutil/src/main/native/cpp/Synchronization.cpp

gaborszita, 2023-02-25:
@virtuald I think this is fixed in the WPILib 2023.4.2 release.

Starlight220, 2023-02-25:
There should probably still be a `stop` method for DataLogManager.

PR #5860 (2023-11-04): [wpiutil] Add DataLog and DataLogManager Stop()
Restarting a stopped log results in creating a new log file with fresh copies of the same start records and schema data records.

Also check to see if the file has been deleted or if the log file exceeds 1.8 GB, and start a new one.

Fixes #4625.
closed 2023-11-04 by PeterJohnson