wpilibsuite/allwpilib #4634: Syncing roboRIO flash drive to development PC

allengregoryiv, 2022-11-14:
**Is your feature request related to a problem? Please describe.**
It would be great if there was a convenient way to sync a RoboRio directory to the development PC. Especially with the increase use of logging frameworks being able to quickly access those logs without removing a flash drive from the RIO would be ideal.

**Describe the solution you'd like**
A WPILib pallet command to sync a specific directory or automatically sync a directory like what we do with the deploy folder but in the opposite direction. This could also be used to sync the preferences file between robots or something like that.

PeterJohnson, 2022-11-14:
For data logs in particular, DataLogTool is specifically designed for easy downloading of log files from the robot.  It doesn't have support for other file types or entire directories, but at some point we'd effectively just be recreating FileZilla.

allengregoryiv, 2022-11-15:
Yes editing in build.gradle would work great.