wpilibsuite/allwpilib #4679: Simulation hangs

Crossle86, 2022-11-21:
I have been doing work on our 2022 code in prepartion for 2023 beta. This work involved coding, testing on sim,
coding some more, testing on sim and so on. Recently a problem with the sim has popped up.

When I start the sim, the gui appears and in the timing window I see the FPGA time rapidly incrementing along with
two watchdogs and TimedRobot (I assume these are threads) whose time increments along with the FPGA time. Normally 
these numbers just run upward until the sim is over.

What I an seeing is that at about 10sec after sim start up, the two watchdog and the timedrobot threads STOP 
incrementing the time and when this happens the sim is not responsive. The FPGA continues to increment. Now if I
wait, some seconds later, the three "hung" threads will start running for a few seconds and while they are running 
the sim is responsive. Then they hang again. They run and hang again sporadically until a minute or so into the 
sim run, where they hang and do not run again.

I have tracing available and it shows the code periodic functions running even when the sim is hung. This would
mean the scheduler is not hung, but no input is delivered to the code when the sim is hung. I assume the lack of
input is timedrobot being hung...but how does the scheduler run if timedrobot is hung? 

If I go back to an earlier commit and run the code at that point in time, the sim works correctly.

It appears this problem started at the point when I set my branch back to an earlier commit (using sourcetree) to 
start over working at that point. 

**Desktop (please complete the following information):**
 - WPILib Version: [e.g. 2022.4.1] 2022.4.1
 - OS: [e.g. Windows 10] 10
 - Java version [e.g. 1.10.2] Built in JDK in wpilib (Java 11)

PeterJohnson, 2022-11-21:
Can you provide a link to your code/project/version of the project that demonstrates the issue?

Crossle86, 2022-11-21:
Here is the project on [github](https://github.com/ORF-4450/Robot23B). Note the contents of simgiu-window.json. Now as is, this project sims correctly. If you make no changes to the gui (simgui-window.json not updated) you can rerun the project and it works correctly. If you change the gui causing the simgui-window.json to  be updated (with correct format) when you run next run the sim it will hang. Hang start varys, sometimes within 10s and sometimes 40-50s but is does hang. I have also noted some variance in which threads hang but one or more will and when they do the sim is unresponsive. This is a crazy problem...

PeterJohnson, 2022-11-21:
Thanks.  The contents of simgui-window.json are clearly incorrect due to unresolved merge conflicts that have been committed.  If you just remove simgui-window.json, does the problem still occur?

Crossle86, 2022-11-21:
So the simgui-windows.json file was modified by sourcetree which flagged merge conflicts, though it never told me. Now I realize that this incorrectly formatted simgui-window file is probably being ignored by the sim and the items on the sim gui screen are placed in whatever default order it decides. So I deleted the simgui-window file and sim ran correctly, items placed randomly. Run again it still worked. So with corrupted or missing simgui-window file sim works and works repeatedly until you change the layout on the sim gui screen. When you stop sim after changing gui layout, a new simgui-window file is created. On the next run of the sim, the hang problem occurs. So presence of simgui-window file is driving the problem somehow.

Now the term hang is really not accurate. When the problem occurs, the threads stop accumulating time, but the gui remains active, you can click on things, move items, enable, disable etc. But joystick inputs are not passed to robot code. Robot code is still running as well. So the problem appears to affect the functioning of the threads, suspending them, and when this happens, no inputs passed to robot code.

rzblue, 2022-11-22:
Unfortunately I can't test this on your project - it's missing a local dependency (RobotLib). I can't replicate this problem in either 2022.4.1 or 2023.1.0-beta-4 with my team's 2022 code.

Crossle86, 2022-11-22:
Some new observations. I see that the hang of the timed robot and watchdog threads only occurs after I switch into teleop mode. When the hang occurs, if I switch back to disabled, after about 5 seconds, those threads resume and run without interruption until I switch back to teleop, where after some seconds those threads hang again. I switch back to disabled and after a bit those threads resume. So the issue only occurs when in teleop. I also have a notifier thread in my code and when the hang occurs, typically that thread also hangs...but not always. 

Further, when I make changes to the gui layout, only some changes are retained. I moved widgets around and switched to light style. When I stop the sim and run it again, the widget locations are retained but not the style. In addition, I am using two gamepad controllers, which I assign to joystick ports. If I stop sim and resume, only one gamepad is retained and not always on the same port I had assigned it to on the last run. This happens when I save from the gui menu or just shutdown from VSCode.

Crossle86, 2022-11-23:
So I test these issues  over and over to make sure what I am seeing is repeatable and to try and establish any patterns. Up to now it appeared certain that deleting the simgui-window file would result in a successful sim run. The next run would not be successful if simgui-window was present. Now, this afternoon, the hang problem occurs simgui-window deleted or present. There may not be a connection between the problem and the simgui-window file. There are some other variations in the behavior of the problem but it still boils down to threads from timedrobot and watchdog and threads in my code will just stop after a few seconds of running in teleop. I am now seeing a case where timedrobot and watchdog start running when I  enter teleop (but hang later) but my notifier thread never runs. The notifier used to run and hang in sync with timedrobot. 

As it stands now, any sim run exhibits the thread hang problem. Note that this issue is project dependent as other similar projects (2022 version) sim just fine. I am clueless as to what I have done to this project to bring this problem on.

Also, the issue where sim gui changes were not saved seems to have gone away with the exception of the gamepad to port mapping, which still fails to be saved.

rzblue, 2022-11-23:
I still can't replicate this issue after running the simulation for over two minutes with and without deleting simgui-window.json. What are the specs of the computer you are running this on? Are you running it with gamepads plugged in?

Crossle86, 2022-11-23:
Yeah, I expected as much. Pads are plugged in. 64 bit Windows 10, Intel(R) Core(TM) i5-6200U CPU @ 2.30ghz, 8 gb memory. 

Things have been normal for some time on this PC. The project in question was copied from 2022 project to prepare for 2023 beta. Wanted to do some clean up work first, so copied project is still 2022 at this time. All went well up to the point where I decided the direction I was going was incorrect, so I moved the project backwards by using sourcetree to reset the current branch to an earlier commit. I worked forward from there again doing things differently in terms my code. While I can't be 100% sure, I think this setting the project to an earlier commit is when the problems started. The code looked good when I reviewed it after setting earlier commit and everything seems fine other than bizarre behavior by the sim. And that behavior is getting more and more strange though the hang is still the key problem. I have been working two branches on this project  and they both exhibit the problem though with variations in what happens. In terms of those variations, an example is that early in this episode, the hang would not occur until sim had run for at least 10sec. Now, my threads do not even start when I select teleop. Also, the threads listed in the thread list have been consistent and map to my code in terms of threads (notifiers) I  create. Now, just a few minutes ago, not all of my threads show up in the list. It seems to be a moving target, though if threads are not running correctly in the whole robot code structure under the sim container, I suppose all kinds of strange things could happen.

I can go back to the original 2022 project and it works fine. 

I am out of ideas so I am moving the project to 2023 beta 4 environment to see if the problem happens there.

Crossle86, 2022-11-23:
So I was wrong on my observation of expected threads in the thread list not being present. This was an error on my part. The hang issue persists on the project in question.

The good news is that I have imported this project into a 2023 beta project, and that 2023 project  sim is working correctly.

Crossle86, 2022-12-17:
This problem appears to be gone under 2023 beta so will close this issue.

closed 2022-12-17 by Crossle86