wpilibsuite/allwpilib #4681: Second Game Pad Controller not recognized by Driver Station

Crossle86, 2022-11-22:
I have modified our code from joysticks for operation to Game Pad controllers. I use two Game Pad controllers. This works as expected on sim on development PC. When I go to the driver station PC and plug in the two Game Pad controllers, only the first one plugged in is recognized by the driver station. The PC dings when the second is plugged in but it does not appear on the DS list of USB devices.

**Desktop (please complete the following information):**
 - WPILib Version: [e.g. 2021.3.1] 2022.4.1
 - OS: [e.g. Windows 11] 10
 - Java version [e.g. 1.10.2] 
 - C++ version [e.g. 17]

Crossle86, 2022-11-22:
F310 GamePads as used in FTC. Now if I run DS on my DEV PC, the  usb list shows  no devices. When I plug in the two F310s, both are recognized. Now when I am at my actual DS PC, and run the DS, the usb device list shows all 5 slots as joystick controllers (which is what we normally use). We only use 4 and they always appear at the top of the list when plugged in via a hub (single connection to PC). So I unplug that hub and all the joystick names back to generic un-highlighted names. 

Now I plug in one game pad and it is recognized right away and appears in the usb device list and works. I plug in the second gamepad, I hear Windows chime indicating new usb device plugged in, but this second game pad does not appear in the usb device list.

ThadHouse, 2022-11-22:
Ah. That sounds like you have all 5 joysticks locked. Since the ds only supports 6 joysticks, the first gamepad gets added to slot 6, but there’s nowhere else for the 2nd controller to go, since it can’t go into any of the 5 locked slots. You can unlock them by double clicking on a joystick. Doing this on the 5th unused joystick should be enough to get the 2nd gamepad to show up.

Crossle86, 2022-11-22:
Thanks. I will try that next time I'm at the lab.

closed 2022-11-29 by Crossle86