wpilibsuite/allwpilib #4698: [cscore] Add support for Y8 and UYVY

MrRedness, 2022-11-24:
My team has been trying to get some cameras working through cscore/PhotonVision using raw formats, and we've attempted to add support for UYVY, but all we've gotten is the UYVY camera shows up as supporting BGR and is black and white. We've also attempted forcing UYVY and still have black and white outputs. We have only tested on Windows so far, due to struggles with Linux installation. Additionally, we have attempted to get the Y8 camera working by forcing BGR format (some research shows that these formats are similar) but this also failed. Error messages do not help, as they only say unknown exception in Native Function (through JNI). Unfortunately, we seem to be in over our heads, as native C for handling cameras is a little bit out of our area of expertise. Any help would be appreciated, and the long-term goal would be supporting these formats directly in cscore. Thank you!

Here's what we have so far: https://github.com/MrRedness/photonvision

PeterJohnson, 2022-11-24:
Can you share the output of `v4l2-ctl --list-formats-out` for the camera?

PeterJohnson, 2022-11-24:
Correct.

CrashTestCoder, 2022-11-24:
I'm from the same team
[image]

CrashTestCoder, 2022-11-24:
Yes, we were in windows. We now have Debian running

PeterJohnson, 2022-11-24:
Ok.  Could you also do `--list-formats-ext`?

CrashTestCoder, 2022-11-24:
[image]

PeterJohnson, 2022-11-24:
Thanks.  Shouldn't be too bad to add support.  Out of curiosity (and so I might get the same camera for testing), what make/model camera is this?

CrashTestCoder, 2022-11-24:
This is the camera we've been talking about so far [](https://www.e-consystems.com/industrial-cameras/ar0234-usb3-global-shutter-camera.asp)

We'd also like to get this camera working with Y8 [](https://www.e-consystems.com/industrial-cameras/ov2311-monochrome-global-shutter-camera.asp)
[image]

CrashTestCoder, 2022-11-24:
I had links...
Current camera : https://www.e-consystems.com/industrial-cameras/ar0234-usb3-global-shutter-camera.asp
Other camera: https://www.e-consystems.com/industrial-cameras/ov2311-monochrome-global-shutter-camera.asp

PeterJohnson, 2022-11-24:
GREY should already work, although it's never been tested.  There's no support for Y16 at present though.

PeterJohnson, 2022-11-24:
Potentially Y8 works in Linux but not Windows?

MrRedness, 2022-11-24:
[image]

This is the error message we receive when running the grayscale camera (on linux).

PeterJohnson, 2022-11-24:
Something weird is going on there... you should never get a protocol error.

PR #4700 (2022-11-24): [cscore] Add UYVY image support
Fixes #4698.
closed 2022-11-24 by PeterJohnson