wpilibsuite/allwpilib #4708: n-dimensional slew rate limiter

rzblue, 2022-11-26:
Limit slew rate by interpolating between two vectors

calcmogul, 2024-08-01:
Something like this in MathUtil for Translation2d and Translation3d may work:
```cpp
/**
 * Limits translation velocity.
 *
 * @param current Translation at current timestep.
 * @param next Translation at next timestep.
 * @param dt Timestep duration.
 * @param maxVelocity Maximum translation velocity.
 */
Translation2d SlewRateLimit(const Translation2d& current,
                            const Translation2d& next, units::second_t dt,
                            units::velocity_unit auto maxVelocity) {
  if (maxVelocity < 0_mps) {
    // throw here
  }

  auto diff = next - current;
  auto norm = diff.Norm();

  if (norm < 1e-9_m) {
    return next;
  }

  auto velocity = norm / dt;
  return current + diff * (units::math::min(velocity, maxVelocity) / norm);
}
```

KangarooKoala, 2025-02-27:
Was this issue specifically for N-dimensional vectors or is Translation2d and Translation3d enough for the desired use case?

calcmogul, 2025-02-27:
I can't think of an FRC use case for 4D+ vectors off the top of my head.

KangarooKoala, 2025-02-28:
Makes sense- I was just double checking since the wording was ever so slightly different.

PR #7806 (2025-02-26): [wpimath] Add Translation2d/Translation3d slew rate limiter
Intends to close #4708 
Adds a function in MathUtil which takes an initial Translation(2d or 3d), and a second translation, the time interval between them and a max velocity. The function will return the output translation, which is the translation with the capped speed(if the speed of the translation is less than the maxVelocity, then the translation should be unchanged). 
will mark PR as ready for review when java code is completed