wpilibsuite/allwpilib #4722: Update command-based starter project to represent new idioms

Oblarg, 2022-11-27:
The default command-based starter project still only really orients users towards writing explicit subclasses of `CommandBase`.  We should renovate it to also represent the subsystem factory and static factory patterns, in keeping with the coming changes to the documentation and other example projects.

The changes do not have to be substantial; really we just need to add an example of each other method of creating a command.

We should also change the naming convention to match that of the new example project in https://github.com/wpilibsuite/allwpilib/pull/4460.

Starlight220, 2022-11-27:
I think the docs rework would be a good step in that direction.

Oblarg, 2022-11-27:
for sure; but the default project should change to match the doc rework.  There's no reason to keep giving people a template that only shows one type of command definition.

closed 2022-12-15 by PeterJohnson