wpilibsuite/allwpilib #4739: Update json implementation

PeterJohnson, 2022-11-30:
The current JSON implementation in wpiutil is based on a very old version of nlohmann JSON with lots of customizations to reduce compile time overhead (untemplating, using LLVM containers, moving a lot of code out of line).  This should be updated to the most recent version.  Ideally it would also use upstream_utils, but this is likely quite difficult due to the number of customizations.

sciencewhiz, 2023-07-04:
On Ubuntu 2023.04 with gcc 12.2, the JsonComparisonTypesTest Less test fails. It passes if gcc is downgraded to GCC 11.3. This appears to have been fixed upstream in https://github.com/nlohmann/json/pull/3446

PeterJohnson, 2023-10-24:
Fixed by #5680.

closed 2023-10-24 by PeterJohnson