wpilibsuite/allwpilib #4747: Romi (Java) example code doesn't build

mjhealy, 2022-12-01:
**Describe the bug**
After generating a new instance of the Romi (Java) example project, I attempted a build, which fails out of the box.

**To Reproduce**
Steps to reproduce the behavior:
1. Generate a new "RomiReference" example instance, targeting Java.
2. Kick off a build.

**Expected behavior**
Code should build without issues, since it is an unmodified example.

**Actual behavior**
Errors reported:
```
Build file 'C:\Users\quasi\Desktop\quasics-frc-sw-2015-pre-light-fix.git\trunk\Experimental\2023\Romi-Beta\build.gradle' line: 38

* What went wrong:
Could not compile build file 'C:\Users\quasi\Desktop\quasics-frc-sw-2015-pre-light-fix.git\trunk\Experimental\2023\Romi-Beta\build.gradle'.
> startup failed:
  build file 'C:\Users\quasi\Desktop\quasics-frc-sw-2015-pre-light-fix.git\trunk\Experimental\2023\Romi-Beta\build.gradle': 38: Unexpected input: '{\r\n    useJUnitPlatform()\r\n    systemProperty 'junit.jupiter.extensions.autodetection.enabled', 'true'\r\n}\r\n\r\n}' @ line 38, column 1.
     }
     ^

  1 error

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 828ms

 *  The terminal process "cmd.exe /d /c gradlew build   -Dorg.gradle.java.home="C:\Users\Public\wpilib\2023\jdk"" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - WPILib Version: v2023.1.1-beta-4
 - OS: Windows 11
 - Java version [e.g. 1.10.2]
 - C++ version N/A

mjhealy, 2022-12-02:
It looks like there's an extra "}" on line 38 of the generated build.gradle file.

rzblue, 2022-12-02:
Fixed in https://github.com/wpilibsuite/vscode-wpilib/pull/559

rzblue, 2022-12-02:
What is the error that occurs when trying to start the simulator?

rzblue, 2022-12-02:
Try running "Java: Clean Java Language Server Workspace", that should fix it.

mjhealy, 2022-12-02:
That does seem to resolve the issue.  (At the least, the simulator starts running; I don't currently have the Romi powered up/connected to verify full functionality.)

closed 2022-12-02 by rzblue