wpilibsuite/allwpilib #4748: Option to use pre-installed vscode in installer.

nullfedora, 2022-12-02:
When I have to re-install wpilib after an update it frustrates me that all of my customization is left on the old version, and that it uses extra space to my current vscode installation. I think that a good solution would be for in the installer to ask if you want to add all of the extensions to a pre-existing vscode installation. I have been just installing the *.vsix files into vscode, but it would be more convenient to just have it automagically install them if the option is selected in the installer.

ThadHouse, 2022-12-02:
We're not going to allow installing into a stock VS Code through the installer automatically. We do some other customizations as well, including turning off updates. There have been cases in the past where a VS Code update has broken the WPILib extension. Having a single setup also makes debugging when teams have issues much easier.

However, the customization thing actually gives a decent idea. We could pretty easily import the configuration from a previous years VS Code, which would allow you to hold customizations year after year.

nullfedora, 2022-12-02:
I think that just keeping customization from a previous years installation would be a viable solution, but I am wondering if this would also install the extensions that were separate from the wpilib ones into the new vscode instance.

ThadHouse, 2022-12-02:
Yes it would. I'd just be making a copy of the data folder. Which should work.

Starlight220, 2022-12-03:
Shouldn't this be transferred to the installer repo?