wpilibsuite/allwpilib #4757: [wpimath] ComputerVisionUtil uses CW+ pitch / calculates incorrect transformation

amquake, 2022-12-03:
Commonly used vision solutions (Limelight, Photonvision) report pitch from the camera's 2d image center to target point as clockwise-positive (CW+), meaning positive values go up, but the wpilib coordinate system considers pitch counter-clockwise-positive (CCW+). The `ComputerVisionUtil` class is commonly used with 2d vision target info and currently takes pitch as CW+:

https://github.com/wpilibsuite/allwpilib/blob/13cdc293826392308f76382dc954e7c3fb4d32bb/wpimath/src/main/java/edu/wpi/first/math/ComputerVisionUtil.java#L58-L59

This is only really a problem when that pitch is used with wpilib classes:

https://github.com/wpilibsuite/allwpilib/blob/13cdc293826392308f76382dc954e7c3fb4d32bb/wpimath/src/main/java/edu/wpi/first/math/ComputerVisionUtil.java#L88-L90

The above snippet means the translation is constructed assuming a positive pitch will have a positive z-value, when it is actually the opposite.

Either the lines in the above example should be changed to invert the pitch angle, or the documentation for the class should be updated to consider pitch CCW+.

------

Additionally, I noticed the `estimateCameraToTarget` method makes incorrect assumptions for the transformation rotation:

https://github.com/wpilibsuite/allwpilib/blob/13cdc293826392308f76382dc954e7c3fb4d32bb/wpimath/src/main/java/edu/wpi/first/math/ComputerVisionUtil.java#L115-L135

That is, that the field-to-camera pose angle is assumed to be equal to `new Rotation3d(0, 0, gyroRadians)`. This is not true if the robot-to-camera transformation has non-zero rotation. This transformation should also be passed to this method to determine the camera angle.

amquake, 2022-12-03:
> or the documentation for the class should be updated to consider pitch CCW+
Also consider that the camera-to-robot transformation parameter for some of the methods in the class must use CCW+ pitch anyway.

calcmogul, 2022-12-03:
In general, ComputerVisionUtil should be using NWU convention, so any instances where it doesn't are bugs.

> The ComputerVisionUtil class is commonly used with 2d vision target info and currently takes pitch as CW+

But it was recently updated to deal with 3D in the North-West-Up coordinate system. If it doesn't do that, that's a bug.

> consider that the camera-to-robot transformation parameter for some of the methods in the class must use CCW+ pitch anyway

If that's the case, that's a bug.

If users have data in an alternative coordinate system, they should use the CoordinateSystem class to convert it to NWU before giving it to ComputerVisionUtil functions.

PR #4775 (2022-12-07): [wpimath] Remove broken and obsoleted ComputerVisionUtil functions
The ComputerVisionUtil class was added before AprilTag support was announced. Now that it has, the functions for estimating a pose from range and yaw are no longer needed; it's just better to get the pose directly from the AprilTag.

The coordinate system on some function arguments was confusing or didn't match the NWU convention the rest of the library uses. It's easier to remove the functions now and add them back after they're fixed since the fixes aren't trivial.

The range function was removed because it uses pitch and yaw [... cut]
commit f18fd41a (2022-12-07): [wpimath] Remove broken and obsoleted ComputerVisionUtil functions (#4775)

The ComputerVisionUtil class was added before AprilTag support was
announced. Now that it has, the functions for estimating a pose from
range and yaw are no longer needed; it's just better to get the pose
directly from the AprilTag.

The coordinate system on some function arguments was confusing or didn't
match the NWU convention the rest of the library uses. It's easier to
remove the functions now and add them back after they're fixed since the
fixes aren't trivial.

The range function was removed because it uses pitc [... cut]
closed 2022-12-07 by PeterJohnson