wpilibsuite/allwpilib #4758: edu.wpi.first.apriltag package disappeared in 2023 beta 4

fovea1959, 2022-12-03:
**Describe the bug**
The classes in edu.wpi.first.apriltag are no longer packaged in WPIlibj.

**To Reproduce**
Steps to reproduce the behavior:
1. Install 2023 beta 3.
2. Read API documentation, see edu.wpi.first.apriltag package, get excited, write code to use the package.
3. Upgrade to beta 4.
4. Sée code not build.

**Expected behavior**
Code should build, or developer should see docs on vendor library to use package.

**Screenshots**
None.

**Desktop (please complete the following information):**
 - WPILib Version: 2023.1.1 beta 4
 - OS: Win 10
 - Java version: whatever shipped in WPIlib beta 4.
 - C++ version: na

**Additional context**
None.

calcmogul, 2022-12-03:
They moved to the apriltag artifact, which we forgot to include in the beta 4 installer. We've merged some PRs related to that, but I don't know if it has actually been fixed yet.

fovea1959, 2022-12-03:
I saw the move, figured it was an oversight. We are working around it, just wanted to get it on the teams radar.

sciencewhiz, 2022-12-03:
https://github.com/wpilibsuite/BetaTest/issues/114

PeterJohnson, 2022-12-12:
Yes, this was fixed in beta 5.

closed 2022-12-12 by PeterJohnson