wpilibsuite/allwpilib #4767: Replace references to SpeedController with MotorController

calcmogul, 2022-12-05:
```
[tav@myriad allwpilib]$ rg SpeedController
wpilibc/src/main/native/include/frc/motorcontrol/MotorController.h
26:   * Sets the voltage output of the SpeedController.  Compensates for

glass/src/lib/native/include/glass/hardware/SpeedController.h
11:class SpeedControllerModel : public Model {
18:void DisplaySpeedController(SpeedControllerModel* m);

glass/src/lib/native/cpp/hardware/SpeedController.cpp
5:#include "glass/hardware/SpeedController.h"
15:void glass::DisplaySpeedController(SpeedControllerModel* m) {
21:    ImGui::Text("Unknown SpeedController");

glass/src/libnt/native/cpp/StandardNetworkTables.cpp
16:#include "glass/networktables/NTSpeedController.h"
145:      NTSpeedControllerModel::kType,
147:        return std::make_unique<NTSpeedControllerModel>(inst, path);
152:          DisplaySpeedController(static_cast<NTSpeedControllerModel*>(model));

glass/src/libnt/native/include/glass/networktables/NTSpeedController.h
16:#include "glass/hardware/SpeedController.h"
19:class NTSpeedControllerModel : public SpeedControllerModel {
23:  explicit NTSpeedControllerModel(std::string_view path);
24:  NTSpeedControllerModel(nt::NetworkTableInstance inst, std::string_view path);

glass/src/libnt/native/cpp/NTSpeedController.cpp
5:#include "glass/networktables/NTSpeedController.h"
12:NTSpeedControllerModel::NTSpeedControllerModel(std::string_view path)
13:    : NTSpeedControllerModel(nt::NetworkTableInstance::GetDefault(), path) {}
15:NTSpeedControllerModel::NTSpeedControllerModel(nt::NetworkTableInstance inst,
26:void NTSpeedControllerModel::SetPercent(double value) {
30:void NTSpeedControllerModel::Update() {
42:bool NTSpeedControllerModel::Exists() {

docs/build.gradle
166:    //enable doxygen preprocessor expansion of WPI_DEPRECATED to fix SpeedController docs
```

jlmcmchl, 2022-12-06:
Are we looking to update documentation related to this change as well? There are 20 files in allwpilib matching the regex `speed(\W|\n)*controller`:

[image]

calcmogul, 2022-12-06:
Yea, we should replace "speed controller" with "motor controller" in the docs as well.

PR #4769 (2022-12-06): Replace remaining references to Speed Controller with Motor Controller
Resolves #4767
closed 2022-12-06 by PeterJohnson