wpilibsuite/allwpilib #4802: button method on CommandJoystick generates exception

gcschmit, 2022-12-12:
**Describe the bug**
Invoking the button method on a CommandJoystick objects results in a java.lang.ArrayStoreException exception when running in simulation (only mode tested at the moment).

**To Reproduce**
Steps to reproduce the behavior:
1. create a new project using the WPILib Project Creator (Java -> Example -> Getting Started)
2. add a couple of instance variables:
```
  private final CommandJoystick joystick0 = new CommandJoystick(0);
  private final Trigger[] joystickButtons0 = new JoystickButton[13];
```
3. invoke the button method on joystick0:
```
joystickButtons0[1] = joystick0.button(1);
```
4. the exception occurs regardless of whether I've associated a keyboard with Joystick[0] or not

**Expected behavior**
I expect a valid Trigger object to be constructed and returned. This wasn't an issue in an earlier beta (not sure which version, unfortunately).

**Screenshots**

Error at frc.robot.Robot.robotInit(Robot.java:42): Unhandled exception: java.lang.ArrayStoreException: edu.wpi.first.wpilibj2.command.button.Trigger
        at frc.robot.Robot.robotInit(Robot.java:42)
        at edu.wpi.first.wpilibj.TimedRobot.startCompetition(TimedRobot.java:106)
        at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:343)
        at edu.wpi.first.wpilibj.RobotBase.lambda$startRobot$0(RobotBase.java:413)
        at java.base/java.lang.Thread.run(Thread.java:829)

Warning at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:358): The robot program quit unexpectedly. This is usually due to a code error.
  The above stacktrace can help determine where the error occurred.
  See https://wpilib.org/stacktrace for more information.
Error at edu.wpi.first.wpilibj.RobotBase.runRobot(RobotBase.java:365): The startCompetition() method (or methods called by it) should have handled the exception above.

**Desktop (please complete the following information):**
 - WPILib Version: 2023.1.1-beta-6
 - OS: macOS 12.6.1 (arm)
 - Java version: openjdk 17.0.5 2022-10-18
OpenJDK Runtime Environment Temurin-17.0.5+8 (build 17.0.5+8)
OpenJDK 64-Bit Server VM Temurin-17.0.5+8 (build 17.0.5+8, mixed mode)

**Additional context**
If I change the code to the following no exception is generated:
```
joystickButtons0[1] = new JoystickButton(joystick0.getHID(), 1);
```

ThadHouse, 2022-12-12:
Thats expected behavior. The return from button() is just a `Trigger` and not a `JoystickTrigger`. Array covariance (Storing a derived array type like CommandButton[] in a base type Trigger[]) is extremely fragile and not recommended. Just allocate as   `private final Trigger[] joystickButtons0 = new Trigger[13];`. That will solve the problem and make the error not possible.

gcschmit, 2022-12-12:
That makes sense. Not sure why I didn't have issue earlier. I made the recommended change. Thanks!

closed 2022-12-12 by gcschmit