wpilibsuite/allwpilib #4826: Update created build.gradle to use Junit 5.6.3+ so that existing team JUnit4 tests still function

fovea1959, 2022-12-16:
**Is your feature request related to a problem? Please describe.**
We have existing libraries from previous years that have JUnit4 tests. 2023 beta-6 generates the following snippet of build.gradle:

````
    testImplementation 'org.junit.jupiter:junit-jupiter-api:5.4.2'
    testImplementation 'org.junit.jupiter:junit-jupiter-params:5.4.2'
    testRuntimeOnly 'org.junit.jupiter:junit-jupiter-engine:5.4.2'
````

We would prefer not to rewrite our JUnit4 tests, so we want to use org.junit.vintage:junit-vintage-engine to run our old tests:

````
    testImplementation 'org.junit.jupiter:junit-jupiter-api:5.4.2'
    testImplementation 'org.junit.jupiter:junit-jupiter-params:5.4.2'
    testRuntimeOnly 'org.junit.jupiter:junit-jupiter-engine:5.4.2'

    testImplementation("junit:junit:4.13.2")
    testRuntimeOnly("org.junit.vintage:junit-vintage-engine:5.4.2") {
        because 'allows JUnit 3 and JUnit 4 tests to run'
    }
````

However, builds fail with:

````
org.gradle.api.internal.tasks.testing.TestSuiteExecutionException: Could not complete execution for Gradle Test Executor 1.
        ...traceback...
Caused by: org.junit.platform.commons.JUnitException: TestEngine with ID 'junit-vintage' failed to discover tests
        ...traceback...
Caused by: org.junit.platform.commons.JUnitException: Failed to parse version of junit:junit: 4.13.2
````

Research reveals that versions of JUnit 5 prior to 5.6.3 have this issue. Ref: [https://stackoverflow.com/a/67566548/17887564](https://stackoverflow.com/a/67566548/17887564) and
[https://github.com/junit-team/junit5/issues/2198](https://github.com/junit-team/junit5/issues/2198).

**Describe the solution you'd like**
Have "Create Project" generate a build.gradle with references to JUnit 5.6.3(+).

**Describe alternatives you've considered**
Our alternatives are to simply fix the versions in build.gradle whenever we have a project with tests, or rewrite our tests for JUnit5 (straightforward, but time consuming)

**Additional context**
It appears that the build for wpilibj itself uses JUnit 5.8.2 (ref: [https://github.com/wpilibsuite/allwpilib/blob/main/shared/java/javacommon.gradle#L115](https://github.com/wpilibsuite/allwpilib/blob/main/shared/java/javacommon.gradle#L115), so apparently it's trustworthy?

rzblue, 2022-12-16:
You can remove the dependencies for junit 5- they are there as a convenience for teams.

Starlight220, 2022-12-17:
I don't see a reason not to upgrade for the next release

closed 2022-12-26 by PeterJohnson