wpilibsuite/allwpilib #4827: SimpleMotorFeedforward calculate Ka

jsimpso81, 2022-12-16:
**Describe the question you have.**

Shouldn't the SimpleMotorFeedforward calculate routine multiply Ka term by dt_seconds?   (Otherwise it seems dt_seconds needs to be included in the Ka constant itself.)

The current code is:

  /**
   * Calculates the feedforward from the gains and setpoints.
   *
   * @param velocity The velocity setpoint.
   * @param acceleration The acceleration setpoint.
   * @return The computed feedforward.
   */
  public double calculate(double velocity, double acceleration) {
    return ks * Math.signum(velocity) + kv * velocity + ka * acceleration;
  }

I think it really should be:

  /**
   * Calculates the feedforward from the gains and setpoints.
   *
   * @param velocity The velocity setpoint.
   * @param acceleration The acceleration setpoint.
   * @param dt_seconds The time between calls in seconds
   * @return The computed feedforward.
   */
  public double calculate(double velocity, double acceleration, double dt_seconds ) {
    return ks * Math.signum(velocity) + kv * velocity + ka * acceleration * dt_seconds;
  }

This allows the calculation of Ka from test data as follows
     Known (or estimated or calculated)
                 - Setpoint Rate of Change during test
                 - Speed, speed setpoint, motor demand a time "n"
                 - motor demand at time "m" where speed_m = speed_setpoint_n
  
    Then:
                   Ka = (motor_demand_m - motor_demand_n ) / setpoint_rate_of_change

    Currently this only works if the calculated Ka is multipled by dt_seconds.

Ka can be calculated either way.  It seems that putting dt_seconds in the code and not in the constant might be better.

**Describe the reason for your confusion.**
This question appeared after doing some testing and calculating Ka from collected data on a simulated robot.

**Is your question related to a problem? Please describe.**
You could choose to make this an "issue" rather an question if you agree.  (Of course I could have completely overlooked something here...)

**Additional context**
I didn't look at the other calculate functions (or the C code.

calcmogul, 2022-12-16:
The equation is correct as is. In `u = Ks * sgn(velocity) + Kv * velocity + Ka * acceleration`, `u` has units of volts, `Ks has units of volts, Kv has units of volts per velocity unit (e..g, meters per second) and Ka has units of volts per acceleration unit (e.g., meters per second squared).

It represents a continuous state-space system solved for the input.
```
u = Ks * sgn(velocity) + Kv * velocity + Ka * acceleration
1/Ka * u = Ks/Ka * sgn(velocity) + Kv/Ka * velocity + acceleration
acceleration = -Kv/Ka * velocity - Ks/Ka * sgn(velocity) + 1/Ka * u
dvelocity/dt = -Kv/Ka * velocity - Ks/Ka * sgn(velocity) + 1/Ka * u
```
The discrete version of this system, neglecting when the velocity crosses over zero, is:
```
xₖ₊₁ = eᴬᵀxₖ + A⁻¹(eᴬᵀ − I)Buₖ + A⁻¹(eᴬᵀ − I)Bc
```
where x = velocity, A = -Kv/Ka, B = 1/Ka, T is the timestep duration, u is the input voltage (held constant between timesteps), and c = -Ks/Ka sgn(velocity).

jsimpso81, 2022-12-17:
Sorry to have bothered you.   Thank you for the information.  I found my issue and it had nothing to do with sample time....

closed 2022-12-17 by jsimpso81

auscompgeek, 2022-12-17:
I had trouble reading the discrete version of the equation in @calcmogul's comment, so here it is more readable:

```math
x_{k+1} = e^{AT}x_k + A^{-1}\left(e^{AT} - 1\right)Bu_k + A^{-1}\left(e^{AT} - 1\right)c
```