wpilibsuite/allwpilib #4836: Examples consolidation

Starlight220, 2022-12-20:
There are too many example projects, which causes duplicates as well as the list being too long to be useful. The tags are also not organized enough to be useful. We should consolidate the example projects and change the tags to actually be useful.
The list on [frc-docs](https://docs.wpilib.org/en/latest/docs/software/examples-tutorials/wpilib-examples.html) is outdated anyway.

<details>
<summary>
Currently, we have these examples:
</summary>

- GettingStarted: differential drive, timed auto, Xbox arcade.
- TankDrive: differential drive, double-Joystick tank.
- TankDriveXboxController: same as above, with split Xbox tank.
- ArcadeDrive: differential drive, single-Joystick arcade.
- ArcadeDriveXboxController: same as above, with split Xbox arcade.
- MecanumDrive: mecanum drive, single-Joystick arcade.
- PDPCAN: PDP telemetry monitoring.
- Solenoids: simple hardware example of solenoids.
- Encoder: simple hardware example of a quadrature encoder.
- EventLoop: flywheel PID+FF using BooleanEvents (not command-based).
- Relay: simple hardware example of a relay.
- Ultrasonic: differential drive P control based on an AnalogInput *(this example naming is rather confusing, as it doesn't use the Ultrasonic class!)*
- UltrasonicPID: exact same as above, just using the PIDController class rather than manually multiplying by `kP`.
- PotentiometerPID: similar to above.
- ElevatorTrapezoidProfile: trapezoid profile with a smart motor controller.
- ElevatorProfiledPID: ProfiledPID.
- Gyro: differential drive, single-Joystick arcade drive with analog gyro `kP` for driving straight.
- GyroMecanum: field-oriented single-Joystick arcade mecanum drive.
- HIDRumble: Xbox rumble.
- Mechanism2d: jointed arm on an elevator, put to Mechanism2d.
- MotorControl:  simple hardware example of a motor controller.
- MotorControlEncoder: same as above, with encoder telemetry.
- Gearsbot: traditional command-based robot with autonomous, differential drive with Xbox tank drive, PIDSubsystem elevator, PIDSubsystem wrist, timed control of a claw.
- QuickVision: one-liner publishing of a camera to CameraServer.
- IntermediateVision: basic vision processing with opencv on another thread.
- AxisCamera: same as IntermediateVision, just with an Axis camera.
- Shuffleboard: Shuffleboard API.
- HatchbotTraditional: traditional command-based robot with autonomous, differential drive with split Xbox arcade drive, hatch subsystem actuated by a double solenoid.
- HatchbotInline: inline command-based robot with autonomous, differential drive with split Xbox arcade drive, hatch subsystem actuated by a double solenoid.
- RapidReactCommandBot: inline command-based robot with autonomous, differential drive with split Xbox arcade drive, intake, PID+FF flywheel.
- SelectCommand: use of SelectCommand to choose between PrintCommands
- SchedulerEventLogging: command logging with random commands, and Shuffleboard markers.
- Frisbeebot: inline command-based robot with autonomous, PIDSubsystem flywheel, differential drive with split Xbox arcade drive.
- GyroDriveComands: mixed inline/traditional command-based, differential drive with split PS4 arcade drive, heading control with PIDCommand and ProfiledPIDCommand.
- Swervebot: swerve odometry/kinematics.
- Mecanumbot: mecanum odometry/kinematics.
- DifferentialDrivebot: differential drive odometry/kinematics.
- RamseteCommand: command-based differential drive path following with RamseteCommand.
- DutyCycleEncoder:  simple hardware example of a duty cycle encoder.
- DutyCycleInput: simple hardware example of a duty cycle input.
- AddressableLED: simple example of addressable LEDs, with a unit test.
- DMA: simple hardware example of a DMA usage.
- Armbot: command-based robot, PID+FF ProfiledPIDSubsystem arm, differential drive with split Xbox arcade drive.
- ArmbotOffboard: same as above, with smart motor controller ProfiledPIDSubsystem arm.
- DriveDistanceOffboard: command-based differential drive with smart motor controller TrapezoidProfileCommand and split Xbox arcade drive.
- MecanumControllerCommand: same as RamseteCommand for mecanum.
- RamseteController: non-command-based differential drive Ramsete path following.
- StateSpaceFlywheel/SysId: state-space flywheel control with different methods of model construction.
- StateSpaceElevator/Arm: elevator/arm with state space and trapezoid profiles.
- SimpleDifferenetialDriveSimulation: non-command-based simulation of differential drive.
- StateSpaceDifferentialDriveSimulation: command-based version of above, with RamseteCommand path following.
- Arm/ElevatorSimulation: non-command-based arm/elevator simulation.
- UnitTest: non-command-based intake with unit testing.
- Differential/Mecanum/SwerveDrivePoseEstimator: differential/mecanum/swerve drive pose estimator.
- RomiReference: Romi command-based.
- DigitalCommunication:  simple hardware example of using digital outputs to send data to an arduino or such.
- I2CCommunication:  simple hardware example of an I2C device such as an arduino.

</details>

I have multiple suggestions:
- The various non-command-based mechanism simulations can be pulled into one project with separate "subsystems" for drive, arm, elevator, and flywheel.
- We can remove at least some of the `[Tank, Arcade] X [Joystick, Xbox]` matrix. Split Xbox arcade drive appears many times, so maybe remove the simple example of it and mix up the other instances a bit.
- At least one of Ultrasonic/UltrasonicPID/PotentiometerPID can be removed.
- AddressableLED can be moved into a larger project, maybe give some ideas as to how LEDs are used to indicate robot state.
- UnitTest can be merged into a different example that has an intake (maybe RapidReactCommandBot?)
- The various telemetry examples (CANPDP, Shuffleboard, Mechanism2d) can be merged into larger projects.
- Simple examples of hardware used in larger projects can be removed.

More ideas are welcome.

virtuald, 2022-12-20:
I'm -1 on combining projects (which spans most of your ideas). Combining projects is going to make it harder for newer programmers to understand what exactly is needed for a particular task. IMHO, most examples for new programmers should be concise and exhibit how to do a single thing.

I think grouping the projects better will make navigating the project list easier and support more examples.

Starlight220, 2022-12-20:
Currently there is a lot of repetition in the projects. That isn't good.

> IMHO, most examples for new programmers should be concise and exhibit how to do a single thing.

This isn't practical, especially for more complicated tasks (or anything command-based). It also doesn't allow testing (which we've noted is a problem multiple times).

ThadHouse, 2022-12-20:
I agree with Dustin on combining examples. Its going to make things worse for many of the things we have examples for.

The big issues is we have 2 types of examples. HW examples, and program structure examples. Maybe we should split those into multiple example types, that get showed in different lists in VS Code. HW examples don't make sense to be fully structured programs, and really should just be a single point on how to use an API.

virtuald, 2022-12-20:
> It also doesn't allow testing (which we've noted is a problem multiple times)

RobotPy has a testing methodology that allows testing full robot programs. It could be easily adapted to Java/C++ and alleviate some of that.

Starlight220, 2022-12-21:
> RobotPy has a testing methodology that allows testing full robot programs. It could be easily adapted to Java/C++ and alleviate some of that.

Testing full robot programs isn't the problem. It's more that those examples don't have any logic or anything that can/should be tested.

---

@ThadHouse where do you draw the line between HW examples and structure examples? For example, what would the `AddressableLED` example be? What about the `Mechanism2d` one? What about `MecanumBot`?

What format do you suggest HW examples be in? `Robot`-only programs? Something else?

truher, 2022-12-27:
Hi, just want to indicate my support for @virtuald here: having a large number of small example projects is much better than writing more frc-docs articles.

i agree with @Starlight220 that the question is really about purpose, and i'd propose an answer, which is that the purpose is pedagogical, and that depending on the topic being taught by an example, the size of the "minimum reproducible example" might be small (demonstrate the LED's) or large (demonstrate swerve pose estimation).  the most important thing is that an example *runs* so a student can poke at it.

so i'd suggest going in the other direction: more examples, not fewer, particularly of things that are important but not well represented (e.g. writing simulation code against websockets, or unit testing sequential operations with simulated driver input).  maybe if there were a wishlist, some teams would step up to fill it out.

Starlight220, 2022-12-29:
As there isn't a consensus as for the examples themselves, and it's too late for any changes in allwpilib structure or vscode, I think a first step in the meantime would be organizing the tags.
Having a proper tag system would be helpful in seeing the distribution of what examples we have, ease of finding examples that do something specific, and so on.

<details>
<summary>Currently, these are the tags we have and how many times they are present.</summary>

     18 "Sensors"
     17 "Joystick"
     14 "Actuators"
     13 "Digital"
     11 "State Space"
      8 "PID"
      7 "Robot and Motor"
      7 "Getting Started with Java"
      7 "Complete List"
      7 "Command-based"
      6 "Vision"
      6 "Drivetrain"
      5 "Analog"
      4 "Simulation"
      4 "Complete Robot"
      3 "Trajectory"
      3 "Pose"
      3 "Path following"
      3 "Odometry"
      3 "Motion Profile"
      3 "Mechanism2d"
      3 "Filter"
      2 "Swerve"
      2 "Shuffleboard"
      2 "Safety"
      2 "Physics"
      2 "MecanumBot"
      2 "Mecanum"
      2 "Lambdas"
      2 "Flywheel"
      2 "Elevator"
      2 "Arm"
      1 "Unit Testing"
      1 "Testing"
      1 "SwerveControllerCommand"
      1 "SwerveBot"
      1 "StateSpaceFlywheelSysId"
      1 "StateSpaceFlywheel"
      1 "Romi"
      1 "RamseteController"
      1 "RamseteCommand"
      1 "Ramsete"
      1 "Preferences"
      1 "Pneumatics"
      1 "Model"
      1 "MecanumControllerCommand"
      1 "I2C"
      1 "Gyro"
      1 "FRC Characterization"
      1 "EventLoop"
      1 "DriveDistance"
      1 "Differential drive"
      1 "Dashboards"
      1 "Characterization"
      1 "CAN"
      1 "ArmBotOffboard"
      1 "ArmBot"
      1 "Advanced Java"

</details>

Many of these aren't very helpful.

Starlight220, 2023-01-22:
Now with the new tag system in place, it's easier to see the following distributions and statistics in the examples:
(A full list [here](https://github.com/wpilibsuite/allwpilib/pull/4961#issue-1536062215), I'm bringing some specific info to attention here.)

We have 62 example projects. (C++ has 63, counting the C one). That's a _lot_, especially given how repetitive a lot of them are.

### Broad categories
- 24 "basic robot" examples with a single mechanism in the `Robot` class.
- 5 "complete robot" examples with multiple mechanisms. All of these are also tagged with "command-based".
- 16 command-based examples (ranging from "complete robot" to very small examples showcasing a single API point).
- 12 "hardware" examples (showcase a specific hardware API).
- 7 vision examples (4 are on-RIO vision; 3 are pose estimators).
- 1 Romi example.

Here, the distribution is rather sensible.
24 "basic robot"s are a bit much, but at a third of all examples it's not as bad.
We might want some more Romi examples?

### Mechanisms
- 23 Differential Drive
- 5 Mecanum Drive
- 3 Swerve Drive
- 6 Arm
- 8 Elevator
- 5 Flywheel
- 2 Intake

23 differential drives is _way_ too much, especially since most of them have the `DifferentialDrive` object and not much else. The correlation with the amount of "basic robot"s is obvious.
Given the rising popularity of swerve, I'd say we should have more swerve examples, though without vendor code it's a challenge.

### Telemetry
- 9 SmartDashboard
- 3 Shuffleboard
- 0 Sendable

For the dashboard that most people use, the Shuffleboard API is majorly underrepresented.
We also have absolutely zero examples of the dashboard-agnostic Sendable API.

### HID
- 22 XboxController
- 17 Joystick
- 2 PS4Controller

Shouldn't be too difficult to replace some of the XboxController usages with PS4Controller.

---

Any ideas? Comments?

auscompgeek, 2023-01-23:
> Shouldn't be too difficult to replace some of the XboxController usages with PS4Controller.

IMO keeping the XboxController usages is fine. The vast majority of teams are more likely to have an XInput-compatible gamepad over a DualShock 4, since the Logitech F310 is in the rookie KoP.

Starlight220, 2023-01-23:
Are the mappings on the F310 identical to Xbox? If not, the benefit the Xbox class brings to those teams is limited; maybe there's reason to add a class for the F310.

auscompgeek, 2023-01-24:
In XInput mode the button mapping on the F310 is identical to an Xbox 360 controller.

auscompgeek, 2026-10-13:
Now that we have a split of examples and snippets, is there anything we want to do here any more?