wpilibsuite/allwpilib #4844: Build system not finding edu.wpi.first.apriltag package

CarloWoolsey, 2022-12-23:
The build system is unable to find the apriltag package located at `edu.wpi.first.apriltag`. 

I'm currently writing a wpilibj example which requires the edu.wpi.first.apriltag package. 
Running `./gradlew wpilibjExamples:build ` from the allwpilib root folder returns returns the following error: 
`~/allwpilib/wpilibjExamples/src/main/java/edu/wpi/first/wpilibj/examples/differentialdriveposeestimator/Drivetrain.java:38: error: package edu.wpi.first.apriltag does not exist`

**Desktop (please complete the following information):**
 - WPILib Version: N/A
 - OS: MacOs 13.0.1
 - Java version: openjdk version "17.0.4" 2022-07-19
 - C++ version: Apple clang version 14.0.0 (clang-1400.0.29.202)

sciencewhiz, 2022-12-23:
I suspect you need to add `implementation project(':apriltag')` to the wpilibjExamples build.gradle

CarloWoolsey, 2022-12-23:
Yep, that's the issue --I think. 
Adding `implementation project(':apriltag')` and `lib project: ':apriltag', library: 'apriltag', linkage: 'shared'` got rid of the error.

srimanachanta, 2022-12-23:
can this issue be closed?

calcmogul, 2022-12-23:
Not yet. The examples should be able to use the apriltag library, and that's not the case on main.

PR #4845 (2022-12-24): Link apriltag package in wpilibjExamples build.gradle
This PR links `edu.wpi.first.apriltag `package in wpilibjExamples `build.gradle`. 
Resolves #4844
closed 2022-12-24 by PeterJohnson