wpilibsuite/allwpilib #4855: Publish step attempts to use x86_64 doxygen binary on aarch64 host

dellagd, 2022-12-25:
Environment:
Debian Bullseye (aarch64) container

When doing the doxygen step of `./gradlew publish`, gradle pulls down a binary artifact for doxygen which is only x86_64 and tries to use it on the aarch64 host, which fails to run.

Fixed by installing doxygen via apt and then commenting out these lines in docs/build.gradle

```
// doxygen {
//     executables {
//         doxygen version : '1.9.4',
//         baseURI : 'https://frcmaven.wpi.edu/artifactory/generic-release-mirror/doxygen'
//     }
// }
```

A more sustainable solution that does not attempt to pull down the artifact for aarch64 hosts should probably be found. Relying on a host install of doxygen seems like an easy route since doxygen does not publish binaries for aarch64.

PeterJohnson, 2022-12-25:
Maybe just wrap this in an if platform block?

sciencewhiz, 2023-01-07:
Fixed by #4862

closed 2023-01-07 by sciencewhiz