wpilibsuite/allwpilib #4860: Add minimalistic "advanced" command-based template

Oblarg, 2022-12-26:
The current command-based template is very explicitly oriented towards beginning teams, and has some expository comments and stubbed classes/methods that most users end up deleting.

For advanced users who want to spin up a command-based project with no frills, there should be an "empty" template that has the bare minimum required project structure with no stubs (i.e., `Robot` and `RobotContainer`, but with nothing in the latter except empty method definitions).

srimanachanta, 2022-12-26:
Should this template keep a lot of the comments/documentation of the standard template? I mostly delete 90% of it.

Oblarg, 2022-12-27:
this should delete basically everything from the standard template except for the scheduler calls and the skeleton of `RobotContainer`.  Comments do not belong in this template; it's supposed to save effort, and comments have to be deleted.

KhiemGOM, 2022-12-27:
I actually leave most of the default codes in Robot as it is because I really did suit my need well but the documentation definitely is redundant. So maybe made it different components in the template so that we have flexibility?

srimanachanta, 2022-12-28:
> So maybe made it different components in the template so that we have flexibility?

It's basically all skeleton so if someone doesn't want to use `RobotContainer`, they don't have to and can just delete the file. The goal of the skeleton template is that most advanced teams end up just removing as much as possible from it so there is no reason to add something redundant that they will just end up removing from it in the end.

KhiemGOM, 2022-12-29:
> > So maybe made it different components in the template so that we have flexibility?
> 
> The goal of the skeleton template is that most advanced teams end up just removing as much as possible from it so there is no reason to add something redundant that they will just end up removing from it in the end.

Well but not all teams fall into two distinct category of beginner and advance. I don't need documentation about everything and example command and subsystem but I also doesn't deviate from the 'robot container' they have already.

srimanachanta, 2022-12-29:
> Well but not all teams fall into two distinct category of beginner and advance. I don't need documentation about everything and example command and subsystem but I also doesn't deviate from the [...]

I am a little confused about what you mean. This is just a second template similar to how TimedRobot features two templates. This one has the RobotContainer class and if someone doesn't want to use it, they can simply delete it. By using this template, they have to delete significantly less fluff from the template before the begin using it.

PR #4861 (2022-12-31): Skeleton commandbased template
closes #4860

Also added `Exit` methods to Robot because they are sometimes used by more advanced teams.
closed 2022-12-31 by PeterJohnson