wpilibsuite/allwpilib #4876: ADIS16470.java does not have a rotation2d method

wud2nuq, 2022-12-29:
**Is your feature request related to a problem? Please describe.**
 The ADIS16470_IMU class does not extend the Gyro class, and so does not have a getRotation2d() method as a gyro would.

**Describe the solution you'd like**
 Provide a getRotation2d() method for the ADIS16470_IMU object.

**Describe alternatives you've considered**
I created a getRotation2d() method within my subsystem class but have not tested it yet.

calcmogul, 2022-12-29:
The ADIS classes don't implement the Gyro interface because Gyro is intended for 2D rotations, but the ADIS IMUs are 3D. For them, we could add a getRotation3d() function that returns a Rotation3d constructed from the internal quaternion.

PeterJohnson, 2022-12-29:
We could also add per-axis getters like getXRotation2d?

calcmogul, 2022-12-29:
The main Rotation2d users care about is yaw, which you can get by calling toRotation2d() on the Rotation3d.

sciencewhiz, 2022-12-29:
Both NavX and Pigeon are 3d but extend Gyro

calcmogul, 2022-12-29:
Then those vendordeps are doing the wrong thing.

Starlight220, 2022-12-29:
> The main Rotation2d users care about is yaw

... which is why 3D gyros implementing Gyro makes sense. It doesn't matter for the few teams who use axes other than yaw, and very much helps the overwhelming majority that cares only about yaw.

calcmogul, 2022-12-29:
The other problem with Gyro is it doesn't support units, which is important for C++.

calcmogul, 2022-12-29:
If we were to fix up the interface, I'd recommend Gyro2d and Gyro3d that are unit-aware. I think we should just deprecate it though.

rzblue, 2022-12-29:
For now we can do what Peter  suggested and add `getRotation` methods for each axis, which would solve the root issue.

binex-dsk, 2022-12-29:
> We don't need dynamic polymorphism (something an interface provides)

Interface-based polymorphism makes it significantly easier to implement base classes for different subsystems (such as drivetrains) and extend it from there. There is still a use case there (which I personally do).

calcmogul, 2022-12-29:
> Interface-based polymorphism makes it significantly easier to implement base classes for different subsystems (such as drivetrains) and extend it from there.

Not necessarily; it depends on what you're doing with it.

First, you may be using a pure interface. Those don't save you any typing, and they mainly act as a safeguard for typos in the derived class. That hasn't helped WPILib in practice because (1) tests and examples will fail to compile if there are API differences, and (2) API mistakes like that are easy to spot early in code review.

Second, you may be using a mix-in. Those let you deduplicate code by moving common behavior into a base class function. You don't have to use inheritance to model that though, and it's generally cleaner and more maintainable to prefer composition over inheritance (that is, a has-a relationship instead of an is-a relationship).

The main use case for inheritance is to perform dynamic dispatch from a base class pointer when you can't know the derived class's type beforehand. An example of that would be the DifferentialDrive drive class taking MotorController references in its constructor.
```cpp
frc::XboxController controller{0};
frc::VictorSPX leftMotor{0};
frc::VictorSPX rightMotor{1};
frc::DifferentialDrive drive{leftMotor, rightMotor};

drive.ArcadeDrive(-controller.GetLeftY(), -controller.GetRightX());
```
However, it's more composable to call the inverse kinematic helper functions directly and pass the results to the derived motor types.
```cpp
frc::XboxController controller{0};
frc::VictorSPX leftMotor{0};
frc::VictorSPX rightMotor{1};

auto [l, r] = frc::DifferentialDrive::ArcadeDriveIK(-controller.GetLeftY(), -controller.GetRightX());
// Other transformations here if you want like acceleration limiting
leftMotor.Set(l);
rightMotor.Set(r);
```

Dynamic dispatch isn't the only way to do polymorphism. Consider [static dispatch](https://en.wikipedia.org/wiki/Static_dispatch) such as [CRTP](https://en.wikipedia.org/wiki/Curiously_recurring_template_pattern) or function overloading. In both cases, you can write the same user code, and the compiler determines which function to call at compile time via the object declaration (CRTP) or the function argument types (function overloading). Dynamic dispatch has performance overhead too (altho Java tries to minimize it), so only use it if you need it.

Object-oriented programming makes it really appealing to model the world in terms of abstract classes, but that isn't the only paradigm for solving programming problems, and it's not the best in every situation. WPILib started as a heavily object-oriented library, but we've trended toward functional programming style and data-driven design because it helped minimize boilerplate and reduce API coupling. We still uses classes, but only if there's mutable state to preserve; otherwise, we use functions. (You can think of a class like a bundle of lambda functions operating on a common closure where the closure contains the class's member variables.)

binex-dsk, 2022-12-30:
> You don't have to use inheritance to model that though, and it's generally cleaner and more maintainable to prefer composition over inheritance (that is, a has-a relationship instead of an is-a relationship).

I don't think that's the case. To be clear, the relationship I'm modeling is a "drivetrain" class that has the base methods necessary for a drivetrain (gyro methods, drive methods, auton methods, etc.) A differential drive train doesn't "have" a drivetrain, it *is* a drivetrain.

In terms of the Gyro, what exactly would the replacement be for the interface? In my implementation, the Drivetrain class has a Gyro object and the base methods like getRotation2d, etc. so there's no need to put it in all the subclasses. When the subclass is initialized, an instance of Gyro (for example, AHRS or WPI_Pigeon2) is passed into the superclass and then is used for all the gyro methods within it. How would this be replaced with composition?

> Consider static dispatch such as CRTP or function overloading.

Those don't exist in Java do they?

calcmogul, 2022-12-30:
> the relationship I'm modeling is a "drivetrain" class that has the base methods necessary for a drivetrain (gyro methods, drive methods, auton methods, etc.)

If you only have one drivetrain, then an interface is pointless boilerplate.

> In terms of the Gyro, what exactly would the replacement be for the interface? In my implementation, the Drivetrain class has a Gyro object and the base methods like getRotation2d, etc. so there's [...]

Instead of using the base class type as the dependency injection argument, just use the derived class type. It's not like you're going to be using more than one kind of gyro at once.

> Those don't exist in Java do they?

Function overloading is a thing. All that is is a function with the same name but different arguments.

binex-dsk, 2022-12-30:
> If you only have one drivetrain, then an interface is pointless boilerplate.

There are several, like Differential and Swerve.

> Instead of using the base class type as the dependency injection argument, just use the derived class type. It's not like you're going to be using more than one kind of gyro at once. Better yet, create the gyro inside the drivetrain subsystem instead of passing it in since the drivetrain is the primary user, and you don't need to mock the gyro behavior yourself (HALSIM does that).

If I'm interpreting correctly, this means having the derived Gyro type in the derived drivetrain itself, and using the gyro methods directly, NOT the superclass.

This does make sense, but I assume that if we need to use the gyro externally, in the base drivetrain class we have the gyro methods (getGyroRotation2d, etc.) and it's implemented according in the superclass according to its gyro's API. Not the end of the world but also not my favorite tbh.

calcmogul, 2022-12-30:
> There are several, like Differential and Swerve.

But your robot only has one drivetrain, and you won't be swapping drivetrain implementations at runtime. A drivetrain interface that's only implemented once just creates more work for yourself, because you have to make the same interface modifications in two places when you're only ever using the derived class directly.

With regard to swapping between differential drive and swerve with one object instantiation in the Robot class: https://en.wikipedia.org/wiki/You_aren't_gonna_need_it. You're only writing one drivetrain subsystem per year, and the mechanical team won't be changing the drivetrain type halfway through the build season.

A robot project has different requirements from an enterprise application in a large company:

| |**Robot**|**Enterprise**|
|-|---------|--------------|
|**Lines of code**|a few thousand|millions|
|**Developers**|a couple|hundreds|
|**Project lifetime**|a few months|decades|

Don't overcomplicate it. You should be focusing on writing the clearest code you can with the minimal amount of boilerplate so that it's easy to make modifications when the robot design team has new requirements. You shouldn't be writing interfaces to save you five minutes of maintenance work once a year.

[image]

> This does make sense, but I assume that if we need to use the gyro externally, in the base drivetrain class we have the gyro methods (getGyroRotation2d, etc.) and it's implemented according in the superclass according to its gyro's API.

Just make a single heading getter in the derived drivetrain class and make the gyroscope instance private. No need to overcomplicate it.

Let's assume the electrical subteam has replaced the gyroscope with a different vendor and you need to update the following drivetrain class to reflect that:
```java
public class Drivetrain {
  // Motor stuff here

  private Pigeon m_gyro = new Pigeon(0);

  public Rotation2d getHeading() {
    return new Rotation2d(-m_gyro.getAngle());
  }
}
```
All you have to do is swap the Pidgeon instantiation for a different gyro, and maybe change the name of the yaw getter in getHeading(). No interface boilerplate required.

Let's assume you want a drivetrain interface anyway though. I can only think of one common property between differential drive and swerve that an interface could encapsulate. The differential drive and swerve drivetrains:

1. Have different motor configurations
2. Have different kinematic configurations
3. Use different closed loop controllers because differential drive is nonholonomic and swerve is holonomic
4. Have a gyroscope

The only common property is a gyroscope, and it's more work to create a "GyroDrivetrain" interface than it is to just instantiate the derived gyro in the derived drivetrain and write a one-liner heading getter for external classes to call.

closed 2023-08-03 by calcmogul