wpilibsuite/allwpilib #4885: Build fails after including DifferentialDrivePoseEstimator.h after winsock2.h

CalebHeydon, 2022-12-31:
It seems like there is an issue with including the `DifferentialDrivePoseEstimator.h` file after `winsock2.h` (only if `_WIN32` is defined) on Windows.  The project fails to build, and it seems like the order that the files are included is important.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new C++ project
2. Add
`#ifdef _WIN32`
`#include <winsock2.h>`
`#endif`
`#include <frc/estimator/DifferentialDrivePoseEstimator.h>`
to any project file.
3. Build the project

**Expected behavior**
The project will fail to compile giving numerous errors referencing a `NumericalIntegration.h` file.

**Desktop:**
 - WPILib Version: 2022.4.1
 - OS:  Windows 11
 - Java version N/A
 - C++ version 17

ThadHouse, 2022-12-31:
This is an unfortunate side effect of including any windows headers. It defines macros for `min` and `max`, which breaks usages of `std:min` and `std:max`. We can work around this in some cases, but generally its HIGHLY recommended to only include windows headers in their own compilation units. There are also a ton of other macros it includes that can break things as well.

closed 2022-12-31 by CalebHeydon

ThadHouse, 2022-12-31:
Just submitted a PR that will fix this specific issue. Basically you can work around it by changing any `std::min(a,b)` call into `(std::min)(a, b)`. We generally try to do this everywhere, but sometimes it gets forgotten.

PR #4887 (2022-12-31): Fix usages of std::min and std::max to be windows safe
Closes #4885