wpilibsuite/allwpilib #4896: What is the release plan / schedule?

jack60612, 2023-01-02:
**Describe the question you have.**
A clear and concise description of what you want clarification on.
How do you plan releases, do you have any specific deadlines, set number of beta's or milestones?
**Describe the reason for your confusion.**
A clear and concise description of why the question requires clarification and what's confusing.
I feel like the release plan should be publicly known so that all teams can stay up to date with changes made and know when to expect a release.
**Is your question related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Additional context**
Add any other context or screenshots about the question here.

calcmogul, 2023-01-02:
We published betas every week or two when there was bugfixes to include or new RIO images to test. Beta is typically from October through December.

We do an initial release up to a week before kickoff to make sure things are ready and so teams can download it beforehand. We've started tagging things already, and we should be done tonight assuming nothing goes wrong.

Note that generated artifacts shouldn't be used until there's been a release announcement. We've pulled releases to fix them before, and installing the correct (less broken) release later is more annoying for the user.

jack60612, 2023-01-04:
> We published betas every week or two when there was bugfixes to include or new RIO images to test. Beta is typically from October through December.
> 
> We do an official release up to a week before kickoff to make sure things are ready and so teams can download it beforehand. We've started tagging things already, and we should be done tonight assuming nothing goes wrong.
> 
> Note that generated artifacts shouldn't be used until there's been a release announcement. We've pulled releases to fix them before, and installing the correct (less broken) release later is more annoying for the user.
> 
> During the season, we make patch releases late in the week every one to two weeks so teams can install it before the weekend.

Thanks for the response, I appreciate everything all of you do!

sciencewhiz, 2023-01-04:
https://www.firstinspires.org/robotics/frc/blog/2023-kickoff-around-the-corner